ra3xdh/qucs_s #608: AppImage link not working

caduser2k, 2024-02-23:
Hi, Could you update/fix the link for the Qucs-S 24.1.0 AppImage download?

thanks very much.

ra3xdh, 2024-02-24:
Yes, I have just noticed that AppImage download link was not available for users without github account. I have uploaded AppImage on the release page to fix this.

Please keep in mind that I am providing DEB/RPM packages for the most of Linux distributions. Arch ad derivatives have its own package in AUR. The AppImage usage is a backup way to install Qucs-S and not recommended except some special cases when no binary package is available. What distribution are you using?

closed 2024-02-24 by ra3xdh

caduser2k, 2024-02-26:
Hi,

thanks for the link update.

In my  personal laptop, I can install the rpm version but at my university account I depend on the AppImage verison.