ra3xdh/qucs_s #611: Error while using File Based Voltage Source

BertheGrandPied, 2024-02-25:
I'm trying to use the File Based Voltage Source (on Debian 12), but impossible to make it work. I tried to simplify at max the .sch, see the picture.

[image]

I have errors "Instance: av1   Message: cannot open file", and my table in the DAT file is not loaded (get message "Warning from checkvalid: vector _net0 is not available or has zero length." that's seems to show that my table is not loaded).

And still get this error, and no table loaded. I tried to change my DAT format (use comma separated, space separated CSV, etc) cant' make it work. Impossible to get it work...

I can't find documentation about this feature, the things I find about QUCS are not working. Is it a bug or I'm doing wrong ? If so, is there a working example ?

Thanks for any clue on that.

And a big thanks for these software which is really nice to use (it's really an improvement compare to other software I used in the past)

tomhajjar, 2024-02-26:
Here is an old Qucs example that runs under Qucs-S.

[File_Data_prj.zip](https://github.com/ra3xdh/qucs_s/files/14398918/File_Data_prj.zip)

ra3xdh, 2024-02-26:
Look for file source example also here: https://github.com/ra3xdh/qucs_s/issues/254

ra3xdh, 2024-02-26:
I have just checked the latest development build and everything works as expected. If nothing above helps then you have probably wrong configuration of XSPICE extensions for Ngspice. If you are using build Ngspice from source this may mean wrong configure options at compile time.