ra3xdh/qucs_s #616: Subcircuit symbol contains overlay of "schematic" and symbol when placed on schematic

tomhajjar, 2024-02-28:
While making a dual-gate MOSFET subcircuit, a problem I have never seen before occurred. The model "schematic" and dual-gate MOSFET symbol are overlaid when placed on schematic. If I place the device on a blank schematic, there are no "extra" lines. 

[image]

[image]

[BF998_MOSFET_prj.zip](https://github.com/ra3xdh/qucs_s/files/14427387/BF998_MOSFET_prj.zip)

ra3xdh, 2024-02-28:
@tomhajjar , I have just tried to recreate the device from scratch using SPICE file and I cannot reproduce the problem. Everything works as expected. I have also inspected my projects containing subcircuits made from SPICE file. Everything works as expected too.  I suspect the schematic file became broken by unknown reason. I sucpect this may be related to the recent schematic class refactoring. @wawuwo , @iwbnwif please check and provide a fix if you can reproduce this. 

Also please keep in mind that Qucs-S contains BF998 model since v0.0.23 release. You don't need to recreate it. I don't consider the update of this library. Other dual gate MOSFETs like BF961 could be added, but I didn't find the SPICE models for them.

ra3xdh, 2024-02-28:
@tomhajjar You have incorrect port number `Num` property. Set the number here instead of `G2` to fix the issue.  This property must be set only to integer numbers. Otherwise you will get the broken schematic. Unfortunately there is no possibility to add a validator for this property. I don't recommend changing automatically assigned port numbers. But port names could be changed. 
[image]

tomhajjar, 2024-02-28:
Thanks for the catch. I was porting an LTspice project by Gunthard Kraus over to Qucs-S and forgot BF998 was native. I also wanted to add a dual-gate device to schematic_symbols.sch.

The LTspice example deletes the C=/L= entries in BF998.cir. ngspice doesn't say in the manual this format is acceptable but it must be.

[image]

[image]

[BF998_MOSFET_prj.zip](https://github.com/ra3xdh/qucs_s/files/14434000/BF998_MOSFET_prj.zip)

tomhajjar, 2024-02-28:
Here is Bordodynov's dual gate mosfet library. I don't think any major manufacturer makes dual gate mosfets anymore.                              Hobbists can still buy surplus.

[Dual-gate MOSFET.zip](https://github.com/ra3xdh/qucs_s/files/14435147/Dual-gate.MOSFET.zip)

tomhajjar, 2024-02-28:
I found the s-parameters for the BF998 and they compare favorably with the Spice model. I will update the project as it expands.

[image]

[BF998_MOSFET_prj.zip](https://github.com/ra3xdh/qucs_s/files/14436445/BF998_MOSFET_prj.zip)