ra3xdh/qucs_s #620: AudioIC tda2003 element does not work correctly

egrast, 2024-02-28:
[image]

tomhajjar, 2024-02-29:
@ra3xdh

I found three versions Jim Thompsons TDA2003 Spice model. The version in Qucs-S has one change, "5" versus "5A" the version from Jim's website. The version Bordodynov supplies is either older or he modified it.

[image]

egrast, 2024-02-29:
@tomhajjar

Thank you !   Works !

I tried to create a model based on "Analysis of TDA2003 and TDA2005 Integrated Amplifiers.pdf".
It is not intended for accurate AC analysis.

[image]

[tda2003-pdf.zip](https://github.com/ra3xdh/qucs_s/files/14449737/tda2003-pdf.zip)

tomhajjar, 2024-02-29:
Nice work. Jim Thompson was an IC designer for Motorola and many others. He created a lot of Spice models as well. His archived website might not last too long.

https://web.archive.org/web/20180806223743fw_/http://www.analog-innovations.com

https://web.archive.org/web/20180806223743fw_/http://www.analog-innovations.com/subcircuits.html

egrast, 2024-02-29:
@tomhajjar

Thank's

tomhajjar, 2024-02-29:
Updated TDA2003. Added LM386, TDA2030 single ended and TDA2030 bipolar.

[image]

[AudioIC_prj.zip](https://github.com/ra3xdh/qucs_s/files/14452479/AudioIC_prj.zip)

ra3xdh, 2024-03-01:
> What is the reason for 5 versus 5A

Ngspice treats this as 5 attoamper. It is near zero.

ra3xdh, 2024-03-01:
@egrast The description of the TDA2003 device says that the LTspice compatibility setting is required for this model. The compatibility mode could be set using spiceinit: https://github.com/ra3xdh/qucs_s/pull/26 I failed to port this model to pure Ngspice and decided to leave it as is. Please always read the device description.

egrast, 2024-03-01:
@ra3xdh 

Ok !
Thanks for the answer.

ra3xdh, 2024-03-01:
@tomhajjar 

> Jim Thompson was an IC designer for Motorola and many others. He created a lot of Spice models 

Thanks for sharing the link. I have grabbed the models from archive to my HDD. It would be nice to wrap some of the models in the Qucs library.

closed 2024-03-02 by egrast