ra3xdh/qucs_s #621: Wrong Github download links of Qucs-S Windows installer and Windows portable package

3813127458, 2024-02-29:
The download links for the Qucs-S Windows installer and Windows portable package do not work:

[image]

[image]

[image]

tomhajjar, 2024-02-29:
https://github.com/ra3xdh/qucs_s/releases/tag/24.1.0

tomhajjar, 2024-02-29:
Also the Windows install instruction at bottom of the page are not necessary and limit having multiple versions of ngspice/xyce spice as well. 

When debugging issues I have to confirm the problem isn't caused by the latest Qucs-S or ngspice version.

[image]
[image]

ra3xdh, 2024-03-01:
I have fixed the package file names. Closing.

closed 2024-03-01 by ra3xdh