ra3xdh/qucs_s #623: AC Trans FFT revisited

tomhajjar, 2024-03-01:
Having issues producing an FFT with a log scale when doing AC/Trans/FFT. It works doing Trans/FFT.

Using a Nutmeg equation isn't working unless I delete the AC analysis.

Using "Logarithmic left Axis Grid" ONLY works maybe 10% of the time if I accidently find the "magic" FFT BW/dF settings.

[image]
[image]
[image]

[AC_Tran_FFT_prj.zip](https://github.com/ra3xdh/qucs_s/files/14462924/AC_Tran_FFT_prj.zip)

ra3xdh, 2024-03-01:
> Using "Logarithmic left Axis Grid" ONLY works if I accidently find the "magic" combination of "settings"

It is not a bug. FFT data from Ngspice may contain negative noise values. This is not allowed for Log axis. I cannot fix this on Qucs-S side.

ra3xdh, 2024-03-01:
> Using a Nutmeg equation isn't working unless I delete the AC analysis.

Everything should work now. @ivandi69 has recently fixed the AC/FFT.

ra3xdh, 2024-03-01:
> Using a Nutmeg equation isn't working unless I delete the AC analysis.

You have a wrong syntax in Nutmeg block. Use short syntax `db(v(out))` and everything will work fine.

closed 2024-03-01 by ra3xdh