ra3xdh/qucs_s #624: Regression Xyce parametric sweep

dwarning, 2024-03-02:
I am not able to plot results from Xyce parametric sweeps. This seems a regression because I see from older example file that it was possible in the past.
I made certain experiments with file format std and raw without success.
[image]

[BSIM6_xyce_prj.zip](https://github.com/ra3xdh/qucs_s/files/14469866/BSIM6_xyce_prj.zip)

tomhajjar, 2024-03-03:
Project works with Qucs 0.0.24. Doesn't work for 24.1.0...

[image]
[image]

dwarning, 2024-03-03:
Latest functional version for Xyce parametric sweep simulation is 2.1.0. It's a pity that this interface has gone.

ra3xdh, 2024-03-04:
Fixed by #627

PR #627 (2024-03-04): Fix #624 (tentative)
This PR should fix #624.
I've never used Xyce and don't have it installed, so only the netlist generation was tested.
commit 5a4d4068 (2024-03-04): Fix #624 (tentative)
commit d17ca1c8 (2024-03-04): Merge pull request #627 from ivandi69/624-regression-xyce-parametric-sweep

Fix #624 (tentative)
closed 2024-03-04 by ra3xdh