ra3xdh/qucs_s #631: Useless black window appears after DC bias calculation

TheTesla, 2024-03-07:
If I press F8 - calculate DC bias... - a black window appears, that an extra click to be closed.

I believe a simulator console output is printed normally there.

I recommend adding a dockable console window, which show all simulator (and other) console outputs to the GUI. Popups are hated by users, especially, if they block the interaction until they are click away on the X.

tomhajjar, 2024-03-07:
Not happening to me in Win 10. Post your file.

ra3xdh, 2024-03-08:
I am regularly testing Qucs-S on different platforms and never seen black windows on F8. What is you platfrom? Please also submit a screenshot. This may be related if you picked a wrong ngspice. You should use `ngspice_con.exe` for Windows. The `ngspice.exe` always opens GUI on launch. Also you may have something like global "run in terminal" setting in your system.

ra3xdh, 2024-03-08:
> I recommend adding a dockable console window, which show all simulator (and other) console outputs to the GUI

This task is planned. See #235 But I cannot say when it will be implemented.

ra3xdh, 2024-03-08:
I have never seen popup windows except the simulation dialog itself using Ubuntu. I still cannot reproduce this.

TheTesla, 2024-03-08:
```
stefan  stefan-ThinkPad-P14s-Gen-3  ~/testing/qucs  develop_filtered  .venv-default  /usr/bin/ngspice
******
** ngspice-40 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2023, The ngspice team.
** Please get your ngspice manual from https://ngspice.sourceforge.io/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Mon Jun 12 16:33:44 UTC 2023
******

```

ra3xdh, 2024-03-08:
It's strange. The simulator settings are correct. I never observed this using KDE/XFCE or Windows. It seems you have some GNOME settings for global modal windows that prevents automatic closing of the simulation dialog. Need to wait until someone will provide debugging and fix. I cannot say what exactly prevents simulation log window from closing.

TheTesla, 2024-03-08:
How can I help you? I can prepare a virtual box with everything to be ready for reproduction. May this be helpful?

ra3xdh, 2024-03-08:
I have to install the Ubuntu 23.10 myself into a VM and debug everything using GDB. I am unable to fix this on my main host. Don't expect a quick fix for this. Probably some users of Ubuntu+GNOME may suggest something.

ra3xdh, 2024-03-08:
Also implementation of #235 will resolve this issue. I cannot say when it will be implemented.

ra3xdh, 2024-03-21:
@TheTesla Please try the following patch and report if the black window still appear.

~~~
diff --git a/qucs/qucs.cpp b/qucs/qucs.cpp
index f9c38448..09f5f6fb 100644
--- a/qucs/qucs.cpp
+++ b/qucs/qucs.cpp
@@ -3424,8 +3424,8 @@ void QucsApp::slotSimulateWithSpice()
                 this,SLOT(slotAfterSpiceSimulation(ExternSimDialog*)));
         connect(SimDlg,SIGNAL(warnings()),this,SLOT(slotShowWarnings()));
         connect(SimDlg,SIGNAL(success()),this,SLOT(slotResetWarnings()));
-        if (!TuningMode) SimDlg->exec();
-        else SimDlg->slotStart();
+        if (TuningMode || sch->showBias == 0) SimDlg->slotStart();
+        else SimDlg->exec();
         /*disconnect(SimDlg,SIGNAL(simulated()),this,SLOT(slotAfterSpiceSimulation()));
         disconnect(SimDlg,SIGNAL(warnings()),this,SLOT(slotShowWarnings()));
         disconnect(SimDlg,SIGNAL(success()),this,SLOT(slotResetWarnings()));*/
~~~

TheTesla, 2024-03-21:
Thanks, it worked without any pop-up window now.

ra3xdh, 2024-03-21:
The PR #651 contains the fix.

closed 2024-03-21 by ra3xdh