ra3xdh/qucs_s #638: Issues RC Transient Simulation

jgardona, 2024-03-10:
This simple rc circuit transient simulation cant generate capacitor curve.

Add 0v to capacitor and 1000 points for simulation. Ac simulation works correctly

[image]

[rctrans.zip](https://github.com/ra3xdh/qucs_s/files/14548232/rctrans.zip)

tomhajjar, 2024-03-10:
V1 doesn't change. How would you see the capacitor charge or discharge? 

Use a step.

[image]

ra3xdh, 2024-03-10:
Everything is correct. As I guess, you are trying to apply initial conditions. Ngspice doesn't apply IC automatically. This question was asked and answered several times before: https://github.com/ra3xdh/qucs_s/issues/291 https://github.com/ra3xdh/qucs_s/issues/226 https://github.com/ra3xdh/qucs_s/discussions/317 This requires to set option `InitialDC=no` for TRAN simulation. I don't know how to make this option more visible, because users don't read the description.

closed 2024-03-10 by ra3xdh

jgardona, 2024-03-10:
> Everything is correct. As I guess, you are trying to apply initial conditions. Ngspice doesn't apply IC automatically. This question was asked and answered several times before: #291 #226 #317 This [...]

You are right. Here is the result. In Ltspice I think this is managed under the hood.
[image]