ra3xdh/qucs_s #647: Add GUI for Qucsconv

ra3xdh, 2024-03-20:
The new version of Qucs-S will include QucsatorRF #370. The Qucsconv utility is a part of QucsatorRF. This tool operates only in CLI mode now. It would be good to add a simple GUI to Qucsconv in the *Tools* menu.

tomhajjar, 2024-03-20:
Sent email...

tomhajjar, 2024-03-21:
The present user interface for Qucsconv is not well thought out. A new user wouldn't understand what file entries or drop down menu choices to make. At a minimum the file extensions should have been listed. Also what the drop down menu choices actually mean would help. I don't even know what some of them will do.

Will the new "Qucsconv-RF" still convert Spice files to Qucs schematics? Qucsconv only works for simple Spice files.

[image]

ra3xdh, 2024-03-21:
> The present user interface for Qucsconv is not well thought out. 

I have never seen this dialog window in Qucs before. How to call it? 

> Will the new "Qucsconv-RF" still convert Spice files to Qucs schematics? 

Qucsconv from Qucsator-RF follows the state for Qucsator-0.0.20: https://github.com/qucs/qucsator I may consider adding some extensions in the future, but I cannot promise when exactly and what.

tomhajjar, 2024-03-21:
Project->Import/Export Data. 

It needs a complete makeover to improve clarity and clear out the unneeded utility choices.

Qucs-0.0.20 has the s-parameter converter. It converts qucs.dat files from an s-parameter analysis to an *.s2p file, ?Not use how it handles more than 2 ports.

[image]

ra3xdh, 2024-03-21:
> It needs a complete makeover to improve clarity and clear out the unneeded utility choices

Yes, the existing converter GUI needs improvement. It should be moved to *Tools* menu and allowed to be launched outside projects. Also it should allow only valid input/output pairs.

tomhajjar, 2024-03-21:
The requirement for opening a Project first isn't too bad since it will save a lot of mouse clicks choosing the input file and indicate the directory where the output file is saved. The input file(s) will generally be in a Project file.

Unfortunately in Qucs when you click the "Browse" button it It points to the location where Qucs is installed instead of the chosen Project directory. Qucsconv should present the chosen Project directory first. 

[image]

tomhajjar, 2024-03-21:
I guess I have some documents referencing qucsconv to read...

[image]

tomhajjar, 2024-03-21:
The statements on this "Help" page seems wrong. Isn't "input" and "output" switched?
https://linux.die.net/man/1/qucsconv

"The software aims to support a variety of data formats and currently supports Qucs datasets, SPICE netlists, CITIfiles, Touchstone, ZVR, VCD and IC-CAP model files as **input** data format and CSV files, Qucs netlists, libraries and datasets as **output** data format."

tomhajjar, 2024-03-21:
Using Qucsconv_rf:

Converting *.dat file to Touchstone S-parameter file is working

Converting SIMPLE 2SC3356.cir file to Qucs Library netlist is working AFAIK. What this file is useful for is unknown...

Converting SIMPLE 2SC3356.cir file to 2SC3356.sch is NOT working. This simple Spice file should easily convert into a sch file.

Attached is an old Qucs project. I created NE85633 HF Spice Model.sch using the qucsconv in QucsStudio.

NE85633=2SC3356

[2SC3356 Amplifiers_qucs_prj.zip](https://github.com/ra3xdh/qucs_s/files/14705703/2SC3356.Amplifiers_qucs_prj.zip)

ra3xdh, 2024-03-21:
> Converting SIMPLE 2SC3356.cir file to 2SC3356.sch is NOT working

CIR to SCH conversion was never implemented in Qucsconv. SPICE netlist could be converted only Qucsator netlist, but not to XML schematic.

ra3xdh, 2024-03-21:
The valid pairs for conversion are the following:
~~~
FORMAT: The input - output format pair should be one of the following:
  inputformat - outputformat
  spice       - qucs
  spice       - qucslib
  vcd         - qucsdata
  qucsdata    - csv
  qucsdata    - touchstone
  citi        - qucsdata
  touchstone  - qucsdata
  csv         - qucsdata
  zvr         - qucsdata
  mdl         - qucsdata
  qucsdata    - matlab
~~~

ra3xdh, 2024-03-21:
> I created NE85633 HF Spice Model.sch using the qucsconv in QucsStudio.

I cannot backport features from QucsStudio to Qucs-S without permission of its developer. Even open parts of QucsStudio may be under copyright.

ra3xdh, 2024-03-21:
> Converting SIMPLE 2SC3356.cir Spice file to Qucs Library netlist is working AFAIK. What this file is useful for is unknown...

This works only for files containing modelcards. The example of such file is here: https://ltwiki.org/index.php?title=Standard.bjt Running qucsconv on it will produce a valid library containing transistor models.

tomhajjar, 2024-03-21:
I fully understand however I've converted multiple Spice files to *.sch using qucsconv in Qucs. The biggest problem has been missing sources so I end up using QucsStudio. I would have to dig through old projects. I saved multiple failed conversions.

These schematics are after I clean them up.

[image]
[image]

ra3xdh, 2024-03-21:
>  fully understand however I've converted multiple Spice files to *.sch using qucsconv in Qucs.

Am I understanding correct? Could you post CLI syntax to convert SPICE to SCH XML? Qucsconv from Qucs never supported this feature. There is no way to convert netlist to schematic. I suspect this may be either something undocumented or you can be using  Qucsconv from QucsStudio by mistake.

ra3xdh, 2024-03-21:
I have inspected the sources, the qucsconv version from Qucs-0.0.20 release cannot produce SCH files, but the version from QucsStudio can.

tomhajjar, 2024-03-21:
Using QucsStudio, this is the converted file for the IRFP250 MOSFET looks like after it has been partially cleaned up. 

[image]

ra3xdh, 2024-03-22:
As I wrote before, I cannot backport the features from open parts of QucsStudio without permission of its developer. I will try to contact Michael Margraf and ask about the license status of the open parts of QucsStudio.

ra3xdh, 2024-03-22:
The import dialog redesign was made by #654. Here is the new dialog look:
[image]

tomhajjar, 2024-03-22:
> As I wrote before, I cannot backport the features from open parts of QucsStudio without permission of its developer. I will try to contact Michael Margraf and ask about the license status of the open parts of QucsStudio.

There really isn't a need to convert Spice to Sch. QucsStudio can't convert the majority of Spice models successfully anyway. It was useful to use it for Qucs but not for Qucs-S.

closed 2024-03-23 by ra3xdh