ra3xdh/qucs_s #652: No .END directive found warning

tomhajjar, 2024-03-22:
I seem to be getting a lot more "no .END directive found" warnings. I cannot see anything wrong with "2N5109_pkg.cir". I modified it and the warning continued. Having same issue with other projects. 

[image]

[2N5109_push_pull_amp_qucs_prj.zip](https://github.com/ra3xdh/qucs_s/files/14715236/2N5109_push_pull_amp_qucs_prj.zip)

ra3xdh, 2024-03-22:
This error is a warning and doesn't affect the simulation. I will remove this message from Qucsconv in the next release, because it can mislead users. Don't insert .END directive in the SPICE model, because the SPICE simulation will not work. 

Also please keep in mind that Qucsator_rf is not recommended for simulation of the SPICE circuits and general purpose circuits. I have no intention and time resources to improve the SPICE support in Qucsator in the near future.

ra3xdh, 2024-03-22:
The attached project archive doesn't contain the schematics files and SPICE netlist.  Also I would not recommend using Qucsator to simulate this circuit except only for academic interest. You may have time-domain convergence issues with this schematic. The Qucs-S contians 2N5109 model in the library that is works fine using Ngspice.

ra3xdh, 2024-03-22:
Fixed by #653

closed 2024-03-22 by ra3xdh

tomhajjar, 2024-03-22:
Here is the qucs and ngspice versions of the preamp originally done by Claudio. He sent me the original schematic.

https://www.qsl.net/in3otd/ham_radio/2N5109_PP_pre/2N5109_PP_pre.html

[image]

[2N5109_Amp_prj.zip](https://github.com/ra3xdh/qucs_s/files/14722112/2N5109_Amp_prj.zip)