ra3xdh/qucs_s #655: XSPICE Cores Library revisited

tomhajjar, 2024-03-22:
1. Cores.lib is not Blacklisted in Qucsator mode
2. If the Cores data in Cores.lib is the same for all Cores, how is the test data in Mike Brinson's Test_Cores_prj different for every test schematic? The schematics look identical. 

[image]
[image]

[Test_Cores_prj.zip](https://github.com/ra3xdh/qucs_s/files/14722660/Test_Cores_prj.zip)

ra3xdh, 2024-03-22:
> Cores.lib is not "visually" Blacklisted in Qucsator mode. The lib itself is however.

The simulator compatibility of library devices is not checked in schematic. It is not possible to implement without library format extension and revising of every device in every library. 

> Originally I thought the cores data was all alike but only the H array is.

Yes, the B-array seems to be correct.

ra3xdh, 2024-03-22:
Moving this issue to discussion: https://github.com/ra3xdh/qucs_s/discussions