ra3xdh/qucs_s #670: Add alternative LC cauer filter

michal777, 2024-03-31:
Hi,
I've got a question about possibility to improve cauer filter calculator.
I tried to design low pass filter and I noticed that only "pi type" is available (even if "tee type" is selected) and only the type with parallel LC in series, no shunt series LC.
I need the tee type with shunt series LC and I noticed that it can be easily made from this pi type. The method is to simply draw the schematic, use the capacitances from pi type filter as inductance and vice versa, then scale inductances by Zref^2 and capacitances by 1/Zref^2.
I didn't even do any mathematical derivation, just tested if such dumb conversion would work and it did. Still it would be good to do the maths to prove the concept and to check if other than low pass filters can be changed that way but,

Any chance such conversions could be added to the calculator?

[image]

michal777, 2024-04-01:
More theoretical at this point. Then I try to adjust a planar resonators to match the shunt LCs using OpenEMS, replace the inductors with microstrips and finally get the whole structure. It's totally inaccurate method but in some of cases I can get a filter which is far from ideal but it's the best I can get and possibly good enough.

tomhajjar, 2024-04-01:
I checked Qucs 19 and 20 and you can't choose a T-type Cauer. Either the Tool never had the capability or there is a bug.

ra3xdh, 2024-04-02:
T-Cype Cauer was never presented in Filter synthesis tool. Need to implement an extension.

jcolafrancesco, 2024-04-08:
Any paper that could serve as a cookbook to implement this missing filter?

tomhajjar, 2024-04-08:
Cauer or Cauer-Chebyshev is also called an Elliptic filter. The definitive book on analog filter design by Anatol I. Zverev called "Handbook of Filter Synthesis" contains charts of normalized filter elements. Many authors used Zverev's info in their books. If you Google Zverev and Elliptic or Cauer you should find the info you need.

closed 2024-05-13 by ra3xdh