ra3xdh/qucs_s #672: Redesign File->Open examples menu

ra3xdh, 2024-04-02:
The *File->Open examples* menu opens file manager window pointing to directory containing examples. The current implementation has many disadvantages. File manager may look very similar to file open dialog on some platforms that may mislead user. Also it is impossible to open Qucs schematics from file manager because `sch` extension is often reserved for other applications. I propose to replace the file manager by open dialog when user click *File->Open examples.

tomhajjar, 2024-04-03:
In Windows the *.sch extension isn't a problem using either File->Open examples or File->Open.

[image]
[image]

wawuwo, 2024-04-05:
@ra3xdh 
>by open dialog

You mean [QFileDialog](https://doc.qt.io/qt-6.5/qfiledialog.html)? (That's the one used by "Open file")

ra3xdh, 2024-04-06:
@wawuwo , done

ra3xdh, 2024-04-07:
Fixed by #674

closed 2024-04-07 by ra3xdh