ra3xdh/qucs_s #679: Improve SpiceLibComp device

ra3xdh, 2024-04-10:
SpiceLibComp device was intended for usage of the unmodified SPICE libraries with Qucs-S https://qucs-s-help.readthedocs.io/en/latest/SubLib.html#usage-of-unmodified-spice-libraries The current implementation has many issues that prevents its usage for circuit design. The following improvements must be done:

- [ ] Provide more symbol file patterns in `share/qucs-s/symbols`
- [ ] Provide a simple way for symbol editing: #147 
- [ ] Provide a simple way to assign the SUBCKT pins and symbol pins without wrapping symbol in Qucs subcircuit
- [ ] Fix the `nested subciruit` warning that appears with Ngspice and may cause simulation convergence issues

Help is appreciated here.

wawuwo, 2024-04-13:
@ra3xdh I've made a small patch enabling Qucs-S to save only a subcircuit symbol of a schematic to a file and to load back symbols from such files. Patch reuses `Schematic` object and resulting "symbol files" are in effect stripped usual schematic files (no components, no properties etc.)

Workflow to create a symbol:
1. Create new schematic 
2. Add the required number of ports
3. Switch to symbol editing mode
4. Draw symbol
5. Save symbol by going to `File->Save` as and selecting "Symbol only" type of file

From now on one can load this symbol back using usual "File->Open" dialog, where "Symbol only" type of file has to be selected.

Unfortunately with this patch symbols from `share/qucs-s/symbols` can't be loaded — they lack `<Qucs Schematc … >` tag on first line.

Please check out branch https://github.com/wawuwo/qucs_s/tree/feature-export-symbol in my repo if it sounds interesting, I think it's not worth of opening of a PR yet.

wawuwo, 2024-04-13:
I forgot to mention that SpiceLibComp can use symbol files made this way, but you have to manually put them into `share/qucs-s/symbols`

ra3xdh, 2024-04-13:
> I've made a small patch enabling Qucs-S to save only a subcircuit symbol of a schematic to a file

Yes, this may help to resolve #147 I was thinking about the similar way. The main problem was how to add/remove ports for symbol without schematic. I will try to evaluate your patch in the next week.

> Unfortunately with this patch symbols from share/qucs-s/symbols can't be loaded

It is not a big problem, because only 4 symbols are shipped with Qucs-S now. These four symbols could be easily recreated using new format.

wawuwo, 2024-04-13:
>how to add/remove ports for symbol without schematic

Maybe a solution is to show "port" tool in paintings alongside with arrow, line, ellipse, etc. when in symbol editing mode, so a user would be able to go straight to symbol editing, skipping the "add required number of ports" part. I'll look into it

ra3xdh, 2024-04-19:
@wawuwo 
> I've made a small patch enabling Qucs-S to save only a subcircuit symbol 

I have tested your patch. I have found this a good solution for #145 The following things should be fixed if possible:

* Ports are show without numbers when opening `*.sym` files This makes difficult to identify ports
[image]
* The four existing symbol files must be replaced with files having a new format. It is not a bug deal because we have only 4 of them.
* It's need to provide a mechanic to add/remove symbol ports
* Ideally add a *New symbol* action in *File* menu

I propose the following logic of adding symbol ports. It could be done using the *Add port* button on the toolbar

* If normal schematic is opened it add subcircuit ports
* If normal schematic (SCH file) is opened in symbol mode (F9) the button is disabled, because the ports are assigned automatically
* If symbol (SYM) is opened the button inserts a symbol port

I am not sure if proposed workflow could be implemented. Alternative solution is also welcome. 
* If the symbol is opened

wawuwo, 2024-04-19:
@ra3xdh
>Ports are shown without numbers when opening *.sym files This makes difficult to identify ports

I need some clarification here… Where should these number come from? I see you've added the symbol to SpiceLibComp component and I guess then the port numbers should be the numbers from `.SUBCKT` entry? I mean, if it's `.SUBCKT Thyristor 99 88 77`, then you expect to see 99, 88 and 77 somewhere around the ports on schematic picture? Or do you mean something like 555 timer has? (I'm sorry if my questions sound stupid, I don't have a lot of experience with it)
[image]

ra3xdh, 2024-04-19:
Normally the port number comes from subcircuit ports (Port device inserted from button on the toolbar) property like `Num=1`. I am not sure how the application derives symbol port number from subcircuit port number. But if the symbol opened without attached schematic, the port is drawn without numbers.

wawuwo, 2024-04-19:
Oh god, it clicked, I finally see what port numbers you mean. Will look into, I guess that's because `adjustPortNumbers` applies some "magic" to symbols and it's not called for "*.sym" files

wawuwo, 2024-04-19:
I've added a commit with patch to display port numbers, but unfortunately has huge disadvantage.

Symbol files store ports like this:
```
  <.PortSym -30 -30 1 0>
  <.PortSym 30 -30 2 0>
  <.PortSym -30 30 3 0>
  <.PortSym 30 30 4 0>
```
where numbers are: x coordinate, y coordinate, auto assigned index number, rotation angle. So for *.sym files the only port label we can obtain is its index number plus some static string, there is no more info in file. In contrast, ports in subcircuit symbols (i.e. usual schematics, not *.sym) derive their labels from names of corresponding ports on schematic, so their labels can be of any form.

The only way to circumvent this I think is to change the format in which symbols are stored, to put "freeform" port numbers into files too.

ra3xdh, 2024-04-20:
It would be sufficient to show the automatically assigned port index number. No need to change the port format.

wawuwo, 2024-04-20:
@ra3xdh 
Please, try the same branch with recent commit. I was able to patch here and there to allow adding ports when in symbol edit mode. 

- it's based on your idea of reusing "insert port" tool
- the format in which ports are stored in a file is unchanged
- in symbol edit mode left mouse button click with "insert port" tool activated spawns a dialog asking for port name, then inserts a port
- in symbol edit mode double click on existing port spawns a dialog asking for new port name. This shouldn't work for usual "subcircuit + subcircuit symbol" files.

ra3xdh, 2024-04-22:
@wawuwo 

> Please, try the same branch with recent commit. 

Yes, the recent improvements on this branch makes the new feature acceptable. I have found two minor issues:

* Delete port doesn't work in `*.sym` files. The port deletion is not needed for SCH.
* The port name dialog allows to enter duplicate port names. This may be a problem when attaching symbol to SpiceLibComp.

> "insert port" tool works in symbol edit mode for usual "subcircuit + subcircuit symbol" files too. But ports added this way are automatically removed when file is saved.

This behavior is correct. The symbol ports and SCH ports must be synchronized. The manual ports edition should be allowed only for detached symbols (SYM). Furthermore the port add button may be disabled when SCH file enters symbol editing mode (F9).

ra3xdh, 2024-05-03:
The merging #700 makes possible converting subcircuit symbols to symbol patterns `*.sym` files. The next step is to implement a custom dialog window for SpiceLibComp device that will allow to assign symbol pin and device pin. The extension of SpiceLibComp will be required.

ra3xdh, 2024-05-31:
I propose to implement a custom dialog for SpiceLibComp containing symbol pins assignment table like in KiCAD. It's need to add something like this for SpiceLibComp:
[image]

ra3xdh, 2024-06-12:
The #755 contains SpiceLibComp redesign.

ra3xdh, 2024-06-15:
I have merged #755. It doesn't contain a fix for "nested subcircuit" warning. This will be a separate PR. The update of the symbol pack also will be a separate PR.

ra3xdh, 2024-06-15:
The *New symbol* action has been implemented by #757. It contains some other improvements of the symbol editor.

ra3xdh, 2024-06-19:
See #775 for "nested subcircuit" warning fix.

ra3xdh, 2024-06-20:
The extended symbols set has been added by #776. This PR finishes the works related to SpiceLibComp redesign. Closing as completed. I will open a separate PR if something will be required for new SpiceLibComp.

closed 2024-06-20 by ra3xdh