ra3xdh/qucs_s #680: qucsator not calculating output

goranmarinkovic, 2024-04-10:
I tried to simulate following design and do not get any output.
[image]

The pulse generated is not visible on the out line and as well the cpld line is quiet... this does not look right.
The test files are attached for further investigation.
[breaking_example.zip](https://github.com/ra3xdh/qucs_s/files/14932712/breaking_example.zip)

ra3xdh, 2024-04-10:
It is not a bug. Qucsator never had transient model for MS lines. The MS lines are considered as zero resistnace for all analysis except SP.

ra3xdh, 2024-04-10:
The Qucsator from Qucs-0.0.20 gives the same result.

closed 2024-04-10 by ra3xdh

goranmarinkovic, 2024-04-10:
Sorry I mixed QUCS-S and QucsStudio... there is see:
[image]

ra3xdh, 2024-04-10:
QucsStudio is closed source project not related to Qucs-S and may have incompatible extensions.

goranmarinkovic, 2024-04-10:
It looks like a missing feature... since if it is possible in QucsStudio it should be available as well for Qucs-S.
It would be a pity if such an important feature is missing.

ra3xdh, 2024-04-10:
Adding transient models for MS lines is technically possible if someone will provide equations.

goranmarinkovic, 2024-04-10:
... here we need an RF expert. Anyone in the community able and willing to share?

ra3xdh, 2024-04-10:
I have converted this issue to discussion. Probably someone could look at this and point a solution.