ra3xdh/qucs_s #682: Warnings during run-time

yurivict, 2024-04-10:
```
$ qucs-s
Gtk-Message: 09:32:48.602: Failed to load module "colorreload-gtk-module"
Warning: Qt: Session management error: Authentication Rejected, reason : None of the authentication protocols specified are supported and host-based authentication failed (:0, )
Warning: QObject::connect: No such signal QProcess::error(QProcess::ProcessError) in /usr/ports/cad/qucs-s/work/qucs-s-24.2.1/qucs/extsimkernels/abstractspicekernel.cpp:72 (:0, )
Warning: QObject::connect: No such signal QProcess::error(QProcess::ProcessError) in /usr/ports/cad/qucs-s/work/qucs-s-24.2.1/qucs/extsimkernels/abstractspicekernel.cpp:72 (:0, )
Warning: QObject::connect: No such signal QProcess::error(QProcess::ProcessError) in /usr/ports/cad/qucs-s/work/qucs-s-24.2.1/qucs/extsimkernels/abstractspicekernel.cpp:72 (:0, )
Warning: QObject::connect: No such signal QProcess::error(QProcess::ProcessError) in /usr/ports/cad/qucs-s/work/qucs-s-24.2.1/qucs/extsimkernels/abstractspicekernel.cpp:72 (:0, )
```

Version: 24.2.1
FreeBSD 14.0

ra3xdh, 2024-04-10:
What is your Qt version (3 digits)? I don't observe the mentioned issues using Qt5.15 build. Need to check Qt6. Something may be different in `QProcess` handling. I will provide a fix soon.

ra3xdh, 2024-04-10:
Yes, the `QProcess::error` was replaced by `QProcess::errorOccured` in Qt6 https://doc.qt.io/qt-6/qprocess.html#errorOccurred

yurivict, 2024-04-10:
I use QT6.

PR #687 (2024-04-12): Fix #682 - Warnings during run-time
This PR fixes #682 .
The signal is errorOccurred(QProcess::ProcessError error) since qt-5.6
[Qt5 doc](https://doc.qt.io/qt-5/qprocess.html#errorOccurred)
[Qt6 doc](https://doc.qt.io/qt-6/qprocess.html#errorOccurredl)
commit 1bb28ff5 (2024-04-12): Fix #682 - Warnings during run-time
commit 7010538b (2024-04-12): Merge pull request #687 from ivandi69/682-qt6-qprocess

Fix #682 - Warnings during run-time
closed 2024-04-12 by zergud