ra3xdh/qucs_s #684: support SAX simulation

joamatab, 2024-04-11:
How could we enable [SAX](https://flaport.github.io/sax/) simulations from QUCS?

@nikosavola

[image]

https://github.com/Qucs/qucs/issues/1082

ra3xdh, 2024-04-12:
The adding the support of the new simulation kernel is technically possible. The following things must to be done:

* Implement netlist generator. Look at `AbstractSpiceKernel` or `VerilogAWriter` classes.
* Implement netlist entries for supported devices like `Component::spice_netlist()`
* Add GUI for simulator execution, log, and error processing
* Implement a converter from simulator dataset to Qucs XML dataset

It is sufficient large amount of code and I am unable to do this work myself because of time constraint. The developers of Qucs-S are working on the project in their spare time. I don't consider such massive extensions of Qucs-S in the near years. But if you find a developer that could implement everything mentioned above, I will accept a patch providing SAX support. This may be a good BS/MS thesis project for someone. Summarizing all above, you have to look for a C++ developer for this task, and I can provide a guidance.

ra3xdh, 2024-04-23:
No volunteers has appeared for this task. Moving to Discussion/Ideas. May be converted back to issue and accepted as the development task if someone will take care of this.