ra3xdh/qucs_s #691: Xyce pametric plot

dwarning, 2024-04-15:
The problem is similar like #161 but seems not solved yet.
I try to make an CV plot with Xyce:

[image]

The resulting data file (attached) looks very well and is plotable with gnuplot.
[nmos_cgs_vgs_xyce_script.txt](https://github.com/ra3xdh/qucs_s/files/14974626/nmos_cgs_vgs_xyce_script.txt)

How I can get the curve into Qucs-s?

ra3xdh, 2024-04-15:
This is as the same as #161 You have a set of 1x1 point plots, but not as a smooth plot. I am aware about this problem and there is no workaround without extension of XML dataset format. Qucs plotting engine cannot understand that a set of 1x1 plots is a smooth curve. You may set *Style: Stars* in the diagram properties dialog and you will see the data points.

ra3xdh, 2024-04-15:
Closing as duplicate of #161

closed 2024-04-15 by ra3xdh