ra3xdh/qucs_s #701: Plotted data appears, disappears, appears...

tomhajjar, 2024-04-22:
I don't know when this started happening. I have the Win 10 "latest" release 24.2.

When running simulations plots appear, disappear, appear...etc.

Run Sim, choose plots, everything OK. Save file, run sim and plots disappear. Run sim again, plots gone so delete plots, choose them again and the plots randomly appear and disappear every sim run. If I do enough iterations of sim, erase plots, choose plots, save file... the plots seem to become "persistant".

EKV3_150_inv_sub_x5_TRAN.sch is the most frustrating since there are so many plots to delete and replace... 

[EKV3_xyce_prj.zip](https://github.com/ra3xdh/qucs_s/files/15068540/EKV3_xyce_prj.zip)

ra3xdh, 2024-04-23:
I confirm this bug. It seems this has appeared after refactoring of Ngspice netlister.

dwarning, 2024-04-23:
Can confirm the behaviour with Xyce simulator and latest Qucs-S under Ununtu.

ra3xdh, 2024-06-01:
The fix provided by #731

closed 2024-06-01 by ra3xdh