ra3xdh/qucs_s #706: Order of blocks in control section

dwarning, 2024-04-25:
I got a problem to switch on model self-heating because the statements in Nutmeg script appears in the end of the control section.
[image]

.control
let number_ib = 0
echo "STEP sw1.ib" > spice4qucs.sw1.cir.res
foreach  ib_act 1e-05 2e-05 3e-05 4e-05 5e-05 6e-05 7e-05 8e-05 9e-05 0.0001 
alter IB = $ib_act
dc vc 0 2 0.01
write spice4qucs.sw1._swp.plot v(B) v(C) v(Tj) i(VPr1)
set appendwrite
echo "$&number_ib  $ib_act" >> spice4qucs.sw1.cir.res
let number_ib = number_ib + 1
end
pre_osdi hicumL2.osdi
altermod @hicumL2va[flsh]=1
.endc

Is there a mean to shift the both statements (especially altermod) in the beginning of the control section?

dwarning, 2024-04-26:
Answering myself: The order is dependent from schematic editing order. Place the the Nutmeg script block at first in an new schematic will show the both cmd's at first in the control block,
The problem that the altermod cmd is without effect because of the following reset:
```
.control

pre_osdi hicumL2.osdi

altermod @hicumL2va[flsh]=1

destroy all

reset

```
How can I setup altermod cmd wo following reset?

ra3xdh, 2024-04-26:
There is no way to control the blocks order in Qucs-S. Furthermore the user must not interfere the automatic netlisting. Try using *Include script* or *Spiceinit* to put load libraries. Also you can rewrite the whole simulation as nutmeg script.