ra3xdh/qucs_s #709: Strange simulation results

dsm, 2024-05-01:
I have a ltspice sim. this is the result(expected result).

[image]

and same circuit in qucs-s result is strange normal diode and 1n4007 diode result also different.

normal diode
[image]

1n4007
[image]

cmat21, 2024-05-02:
Hi @dsm , consider checking the number of point on settings transient, maybe that will help.
[image]

[image]

ivandi69, 2024-05-02:
tran 0.001 3
works just fine

ra3xdh, 2024-05-03:
@dsm 

You are using a default diode model in a first circuit. LTspice and Qucs-S have the different default diode model. You have `BV=10V` in Qucs-S by default and diode goes to breakdown producing wrong output. Set the `BV=1000V` and everything will work as expected.  The another issue is that you have used default transient points number. Ngspice is able to adjust transient time step but only to some extension. You have to set the points numbers manually for long transient simulation durations. Espesially if Tstop is greater than 5-10 waves of the input signal source. You have Tstop=3s and F=50Hz. Ngspice may skip the sine waves and show wrong result. 

[image]

The convergence issues with 1N4007 model is not a Qucs-S bug. It is related to Ngspice bug: https://sourceforge.net/p/ngspice/mailman/ngspice-devel/?viewmonth=202205 I have reported this once, but it seems it has appeared again. The workaround is to set `Ikf=1e-20` in the diode parameters. I can patch the diodes libraries to provide this workaround for all users.

dsm, 2024-05-03:
BV=1000V will solve the problem and number of the point not set to default but not as high as 32768. thank you for your explanation.

ra3xdh, 2024-05-03:
I am transferring this to discussion. The issue with `Ikf` parameter probably should be reported to Ngspice team after providing a minimal netlist example to reproduce.