ra3xdh/qucs_s #712: macos build failed with qucsator_rf

dsm, 2024-05-05:
Hi, I'am developing to github action to build mac version of qucs but macos build failed with qucsator_rf. error is related to VERSION file like old issue on windows MSVC build.

OS: Macos 14.4.1 arm64

ra3xdh, 2024-05-06:
It's need to exclude the toplevel directory from include directories list. This directory doesn't contain header files. It's need to look for old commits fixing this issue for Qucs GUI and apply the similar patch for qucsator_rf. 

Are you planning to add Github action for Windows too or for MacOS only? Windows support has higher priority than MacOS.

dsm, 2024-05-06:
I'm thinking of adding it for windows and macos. I can add windows support earlier. 

I've been experimenting with it since i just got a new macbook.

i'm running out of time for this week. i'm thinking of speeding it up in the coming weeks. 

Which installer do you prefer for the windows side installation file (ex. NSIS installer)

ra3xdh, 2024-05-06:
Yes, the Windows Github action would be useful. I am using MSYS2 to build binaries and InnoSetup to generate the installer. Here is my InnoSetup script template: https://github.com/ra3xdh/qucs_scripts/blob/master/qucs.iss

dsm, 2024-05-06:
Thank you, github action has native windows-2022 server runner maybe using for msvc build. msys2 for building mingw variant.

dsm, 2024-05-06:
I think qucsator_rf cmake file need refactoring and flex,bison,gperf,dos2unix,sed all of this needed I don't know why. Can you explain ?

ra3xdh, 2024-05-06:
> I think qucsator_rf cmake file need refactoring

I don't recommend a deep refactoring of qucsator-related CMake stuff. Maybe restrict to fixing warnings from modern CMake and revert autotest stuff: https://github.com/Qucs/qucs-test This work definitely should go to the separate PR from Github actions. 

> flex,bison,gperf,dos2unix,sed all of this needed I don't know why

* Flex/Bison are needed to generate a parsers for different netlists
* Gperf generates some syntax analyzer tables. I don't know why the developers of Qucsator decided to use Gperf for this purpose.
* Dos2unix is needed to postprocess the files generated by gperf. Otherwise build fails on some platfroms. 
* Sed is also needed for postprocessing of the files generated by gperf.

dsm, 2024-05-06:
I am not able to build native Windows 10 build gperf in path but gperfappgen.exe note found error i didn't found this exe. 

cmake minimum maybe be increased to 3.10 and some compiler flag set using modern cmake

ra3xdh, 2024-05-06:
I recommend to use MSYS2 environment to build qucsator_rf. It is deeply integrated with Unix shell and there may be difficulties with build using native Windows shell. `gperfappgen.exe` is build by GCC from the CPP files from gperfappgen.exe. It generates input tables for gperf. The EXE could be found somewhere in build directory.

dsm, 2024-05-06:
I'll try MSYS2 for build, thank you

ra3xdh, 2024-05-06:
Here is Github MSYS2 workflow template: https://github.com/msys2/setup-msys2

dsm, 2024-05-06:
https://github.com/ra3xdh/qucsator_rf/pull/3 will be fixed this issue.

ra3xdh, 2024-05-07:
I have accepted https://github.com/ra3xdh/qucsator_rf/pull/3 and updated qucsator_rf submodule.

closed 2024-05-08 by ra3xdh