ra3xdh/qucs_s #714: Expired Sign Keys

Ladniy, 2024-05-08:
Hi! I am trying to install qucs-s from Debian 12 repo, but got an error with GPG key:

```
W: GPG error: http://download.opensuse.org/repositories/home:/ra3xdh/Debian_12  InRelease: The following signatures were invalid: EXPKEYSIG 184B98BE3E68CD11 home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>
E: The repository 'http://download.opensuse.org/repositories/home:/ra3xdh/Debian_12  InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
```

Any chance to update it?

ra3xdh, 2024-05-09:
I cannot confirm the key expiration. The GPG key is valid until 2026. You can check its status here: https://build.opensuse.org/projects/home:ra3xdh/signing_keys 

I cannot say why Debian doesn't accept keys from OBS. If you have trouble with adding repository download and install the DEB package using `dpkg`. You can find a link for DEB files here: https://software.opensuse.org/download.html?project=home%3Ara3xdh&package=qucs-s

Ladniy, 2024-05-09:
> I cannot confirm the key expiration.

The key that you provide is does not. If i use it then repository is signed up and i can install Qucs-S.
But the key from [OBS instruction](https://software.opensuse.org/download.html?project=home%3Ara3xdh&package=qucs-s) is definitely expired after may 2-nd. See `apt-key list` output:

```
/etc/apt/trusted.gpg.d/home_ra3xdh.gpg
--------------------------------------
pub   rsa2048 2016-11-08 [SC] [expired: 2024-05-02]
      69B1 C091 6FDB 15A8 7417  43E8 184B 98BE 3E68 CD11
uid           [ expired] home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>
```

ra3xdh, 2024-05-09:
It's strange. OBS doesn't provide any means to control the repository keys. I cannot to find a guide how to extent the key, except this guide for the global project key: https://forums.opensuse.org/t/how-to-update-expired-obs-home-project-signing-key/119147/3 Therefore I don't understand how to fix this and I need an assistance with OBS settings.

ra3xdh, 2024-05-09:
I will try to delete the repository and recreate it again. Maybe this will help.

ra3xdh, 2024-05-09:
I have just recreated Ubuntu and Debian repositories. I have tried Ubuntu 22.04 and it accepts key. The `apt-key list` shows the key is valid. Try to remove the recently added repositories and old keys from apt and then add the repository again.

Ladniy, 2024-05-09:
Now the key is updated and successfully signs a repository, many thanks!

The `apt-key list` output:

```
/etc/apt/trusted.gpg.d/home_ra3xdh.gpg
--------------------------------------
pub   rsa2048 2016-11-08 [SC] [expires: 2026-06-28]
      69B1 C091 6FDB 15A8 7417  43E8 184B 98BE 3E68 CD11
uid           [ unknown] home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>

/etc/apt/trusted.gpg.d/home_ra3xdh_old.gpg
------------------------------------------
pub   rsa2048 2016-11-08 [SC] [expires: 2026-06-28]
      69B1 C091 6FDB 15A8 7417  43E8 184B 98BE 3E68 CD11
uid           [ unknown] home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>
```

closed 2024-05-09 by Ladniy