ra3xdh/qucs_s #719: Add Windows CI setup using Guthub actions

ra3xdh, 2024-05-13:
The implementation of the Windows CI using Github actions involves two tasks

- [x] Add basic CI setup to test windows compatibility. This task is implemented via #718 
- [ ] Add Windows snapshots generation procedure using Github actions. This will provide access to the newest snapshot builds for Windows users.

dsm, 2024-05-13:
I'll creating a portable version. running windeployqt6 for all .exe exept qucsator_rf. download and extract ngspice in portable folder. portable release include misc,share folder and these folders where to locate in repo

ra3xdh, 2024-05-13:
The snapshot is intended mainly for testing of the Qucs-S by Windows users that already have the stable version installed. So, it may not include ngspice, because the user already has the ngspice installed and configured. The `misc` subdirectory is needed only for installer generation by InnoSetup and doesn't affect the program operation. The `share` subdirectory is generated automatically after `make install` step.  In the long perspective we may enable generation of the Windows release package from the `master` branch.

dsm, 2024-05-13:
I tried deploy in local MSYS2 env windeployqt6 doesn't all .dll file and doesn't run all exe

dsm, 2024-05-13:
I tried different but just found qt dlls other dlls ex. libfreetype libpng  not found I'll tried qt5 tomorrow

dsm, 2024-05-14:
I tested qt5 build in mingw64 env and doesn't find other dlls output of the command:

```
$ windeployqt.exe qucs-s.exe --no-translations --debug --list mapping
"C:\msys64\mingw64\bin\Qt5Core.dll" "Qt5Core.dll"
"C:\msys64\mingw64\bin\Qt5Gui.dll" "Qt5Gui.dll"
"C:\msys64\mingw64\bin\Qt5PrintSupport.dll" "Qt5PrintSupport.dll"
"C:\msys64\mingw64\bin\Qt5Svg.dll" "Qt5Svg.dll"
"C:\msys64\mingw64\bin\Qt5Widgets.dll" "Qt5Widgets.dll"
"C:\msys64\mingw64\share\qt5\plugins\iconengines\qsvgicon.dll" "iconengines\qsvgicon.dll"
"C:\msys64\mingw64\share\qt5\plugins\imageformats\qgif.dll" "imageformats\qgif.dll"
"C:\msys64\mingw64\share\qt5\plugins\imageformats\qico.dll" "imageformats\qico.dll"
"C:\msys64\mingw64\share\qt5\plugins\imageformats\qjpeg.dll" "imageformats\qjpeg.dll"
"C:\msys64\mingw64\share\qt5\plugins\imageformats\qsvg.dll" "imageformats\qsvg.dll"
"C:\msys64\mingw64\share\qt5\plugins\platforms\qwindows.dll" "platforms\qwindows.dll"
"C:\msys64\mingw64\share\qt5\plugins\printsupport\windowsprintersupport.dll" "printsupport\windowspr
intersupport.dll"
"C:\msys64\mingw64\share\qt5\plugins\styles\qwindowsvistastyle.dll" "styles\qwindowsvistastyle.dll"
```

ra3xdh, 2024-05-18:
I was wrong about `windeployqt`. This tools deploys only Qt DLLs and plugins. I remembered that I had to copy the non-Qt DLLs manually to the `bin` directory to get the Qucs-S distribution operational. I didn't find a straightforward way to deploy the DLLs using MSYS2 environment. You may get the list of DLLs from the binary package and copy them using `cp` command from `mingw64/bin`. Also this script may help: https://blog.rubenwardy.com/2018/05/07/mingw-copy-dlls/ This script could be placed in `contrib` subdirectory and executed at deployment step.

dsm, 2024-05-19:
Objdumb not give all dlls. LDD tool maybe used, I tried manual copy and everything's work fine.

ra3xdh, 2024-05-19:
We can put copy DLLs in script and execute it after windeployqt to automatize the package build. Then the package could be uploaded as the build artifact.

dsm, 2024-05-19:
Yes, we can put DLLs in a script, and i just find another way LDD tools give dlls with their path. I will add a basic method for snapshot release.

ra3xdh, 2024-05-23:
The windows snapshot generation provided by #725. I will put an instruciotn at the Discussion/Announcement  how to download and use Windows snapshots.