ra3xdh/qucs_s #732: the open/save box is very slow

windilt, 2024-06-02:
my system is Arch Linux with KDE Plasma 6.0.5
the open/save box is very slow to appear and lost response before it appear. And the selection action is very slow and easy to lost response too
hope this can be fixed in next version, thanks~

ra3xdh, 2024-06-02:
Sorry, I cannot reproduce this issue. And I cannot fix what I cannot reproduce. I am using Arch(Manjaro) with the latest KDE6+Wayland too and everything works as expected on my machine. Furthermore I have never seen the mentioned issues on other platforms including X11 and Windows. Everything works as expected too. Are you using Wayland? Do you have `qt6-wayland` package installed? This package is not installed by default and the missing package may produce very weird bugs.

windilt, 2024-06-04:
Thanks a lot for your kindly help.
I use X11, because my screen is flashing with wayland, I guess it is linked with my GTX1650 driver. Anyway, I will try wayland in near further, hope it can solve the issue.

ra3xdh, 2024-06-04:
Also try to check the settings in `qt6ct` tool. You may have some effects enabled.

windilt, 2024-06-10:
I try a lot. Finally, after I new create a user, the open/save box is ok now with the new user. So there must be some problem in my setting. I think this issue can be closed. Thanks a lot for your time and help@ra3xdh

closed 2024-06-10 by windilt