ra3xdh/qucs_s #736: How use xxx.va model and other mos transistor which not included in Qucs-s?

LujiuZhang, 2024-06-05:
Dear developer, 

When I used F9 to generate circuit symbol for a verilog-a file and tried to open it, I got an error as shown in the picture. Could you tell me how to solve this problem? thank you. In addition, what is the whole process from building the verilog-a model to generating the symbol, and then designing the circuit using the symbol? I can't find the related tutorial, can you provide it? Another question is, if I want to use the mos model in the pdk file provided by other semiconductor manufacturers, what should I do? I hope you can answer my questions. Thank you.
[image]

ra3xdh, 2024-06-05:
The Verilog-A workflow using JSON files described in MOS-AK presentation from 2014 is deprecated. It is kept only to open old projects. I will add an explicit deprecation warning in the next release. Please use a new workflow with subcircuits and "Z" type device. See #411 for example. Only OpenVAF+Ngspice is supported. Qucsator-RF doesn't support ADMS anymore. The OpenVAF support for Qucsator-RF may be added in a far future.

LujiuZhang, 2024-06-05:
Ok, thank you. One more question, if I draw a mos tube myself, how should I call it? The first picture is the transistor I drew, and the second picture is the netlist file after the call. Obviously, the model was not successfully called, so how do I solve this problem? 
[image]
[image]
[image]

ra3xdh, 2024-06-05:
The #411 contains an example schematic with tunnel diode model. Pleas download it to learn how the subcircuit using VA-model is done.

LujiuZhang, 2024-06-05:
Thank you!  If use "Z" type device when verilog-a model is considered, the shape of device descripted by verilog-a can't be modified, this probelm can be solved?

ra3xdh, 2024-06-05:
#737 adds deprecation warning. Moving this thread to Discussion/Q&A