ra3xdh/qucs_s #739: Selection Box, Selecting Items and Moving Items Issues

tomhajjar, 2024-06-07:
The selection box has radically changed making it unusable on Windows. The mouse Cursor is not even close to the corner of the selection box.

It's almost impossible to select items as well. The cursor has to be placed away from the item you want to select. Trying to select a single items when multiple items are nearby is impossible.

Also when I select an item and try to move it, it jumps far away from the cursor making moving or resizing items impossible.

It's literally impossible to make or edit a schematic, sub-circuit or model.

[image]
[image]

ra3xdh, 2024-06-07:
I can only partially reproduce your bug report. I have tested the both the latest Linux and Windows version. The results are the following:

* Schematic editing, insert, select, move device or wire works as expected
* Placing paintings in schematic mode works as expected
* Placing paintings in symbol editing mode has shifted mouse pointer. See the attached screencast. Linux version is affected too. Only some subcircuits are affected. 

[Screencast_20240607_191017.webm](https://github.com/ra3xdh/qucs_s/assets/4920080/f4d7e493-19a0-4cfa-8d6a-6a722ccef5d1)

I think this may appear after #723 @wawuwo , could you have a look? You may use this schematic containing subcircuit symbol as the test case. 
[Tunnel_new.sch.gz](https://github.com/user-attachments/files/15742358/Tunnel_new.sch.gz)

ra3xdh, 2024-06-07:
> I'm trying to test the new Varactor lib but had to give up.

New Varactor.lib library will work with stable version of Qucs-S. Just copy the library file to the `share/qucs-s/library` subdirectory. You need not to draw symbols for this library. It uses a standard diode model with `Cj0` parameter set according device specification.

tomhajjar, 2024-06-07:
Is there a reason you didn't use the varactor symbol for the default symbol for library and schematic? One is forced to change it when making the schematic.

For a Varactor C versus V curve to be correct either a look up table of C/V needs to be used or "M" can be used for an approximation. Rs should also be added. Look up table requires a script.

Another option is to make a sub-circuit.

 In reality varactors are starting to disappear from designs....
[image]
[image]

wawuwo, 2024-06-07:
Hello guys! I am sorry, that is totally my mistake: I missed the `restore()` call on QPainter in `id_text::paint`.

The id_text is this little thing:
[image]

The lack of `restore()` makes changes to QPainter applied in id_text::paint "stick to" it and because of that everything painted after the id_text is affected (mostly shifted in some direction). The size of "shift" depends on position of id_text. Try move the id_text to see how it works.

I believe #740 should fix the issue.

wawuwo, 2024-06-08:
>Why is the selection box using thick lines?

Just a consequence of refactoring of rendering. It's possible to make it use the same line width despite of zoom level, like this:

https://github.com/ra3xdh/qucs_s/assets/40355531/30ee11e0-1acd-482d-9ba0-4a204d5dd31e

tomhajjar, 2024-06-08:
When I make detailed schematic symbols I use the 1x1 grid. Being able to easily select, move or resize specific items is important. The older Qucs system I never had an issue. I need a new nightly release to check out how well your new system works. 

[image]

[Symbol_creation_prj.zip](https://github.com/user-attachments/files/15746246/Symbol_creation_prj.zip)

ra3xdh, 2024-06-08:
@tomhajjar, You may download the snapshot with fix using the following link: https://github.com/ra3xdh/qucs_s/actions/runs/9423149593/artifacts/1580665163

wawuwo, 2024-06-08:
@ra3xdh what about the selection rectangle border line width? Should it be kept as is or changed to be like in the example [above](https://github.com/ra3xdh/qucs_s/issues/739#issuecomment-2155770403)?

ra3xdh, 2024-06-08:
> what about the selection rectangle border line width?

The selection rectangle board width should not be scaled with schematic. This should be fixed if possible. It the fix requires some deep modifications, the things may be left as is. The fix may go to the separate PR.

tomhajjar, 2024-06-08:
Guys I now have serious problems with my Main and Back-up PCs after downloading the above snap shot. All my folders on my C: drive are locked in Read-Only mode and nothing I do will make them Read/Write.

I had issues with my Main PC after the download so went to my back up. I confirmed back-up folders were OK but once I copied the snap shot files over my network, all the back-ups folders became Read-Only. I'm unsure what to do now.

ra3xdh, 2024-06-08:
This is definitely not related to snapshot itself. The archive dosen't modify the write permissions of the folders. I am regularly testing both Windows 10 and 11 and everything works as expected.

ra3xdh, 2024-06-08:
The selection rectangle border width fixed by #741 (merged)

wawuwo, 2024-06-08:
@tomhajjar It's sounds very strange (almost impossible) that a mere downloading a file turned off the write permission of some *specific* folder. I believe this wasn't the first when you updated the app from a downloaded snapshot-installer, maybe something had changed since the last update? Did you use the same user as before? Anyway it's hard to tell anything without you providing more details, tracing your steps

ra3xdh, 2024-06-08:
> I now have problems with my Main and Back-up PCs after downloading the above snap shot. Windows will not let me write the files to the QUCS-S 24.2.0 folder. 

I have just tried to download an archive from the link in this thread and extract it in some folder using physical Windows-10 installation. Everything works as expected. You shoudl allow to run executable `qucs-s.exe` from unknown source. FS write permission is not locked. The most probably this problem is on PC side.

tomhajjar, 2024-06-08:
There was definitely something wrong with the last download using the link above. The latest download from the link below doesn't have this problem but unfortunately selection box and cursor positions are still wrong as shown below. 
https://github.com/ra3xdh/qucs_s/releases/tag/continuous_build

Is there another download link I should use?

[image]

tomhajjar, 2024-06-09:
Another issue I'm having is some of the lines used to make the subcircuit "diode" symbol are in the wrong position on old subcircuits I made. See the two subcircuits in the attached project. They look normal on older Qucs-S versions.

If I click on the "SUB" parameters and move it, the lines making the "diode" symbol jump back where they are supposed to be.

If I select a line, the line jumps to another location even though I only selected it.

[image]
[image]
[image]

[Varactor_ngspice_prj.zip](https://github.com/user-attachments/files/15750436/Varactor_ngspice_prj.zip)

tomhajjar, 2024-06-09:
As I go through old projects with subcircuits I'm finding more symbols that are "corrupted". Some of the lines and text are displayed wrong. Something is wrong with Qucs-S. It doesn't work properly displaying or editing symbols and the selection box is offset from the cursor..

Attached is an old project both Vadim and I discussed in the past. The "genopa" subcircuit symbol is screwed up. If I select "SUB" and move it, the symbol displays correctly.

[image]
[image]

[OpAmp_Sub_Param_Pass_prj.zip](https://github.com/user-attachments/files/15750448/OpAmp_Sub_Param_Pass_prj.zip)

ra3xdh, 2024-06-09:
> There was definitely something wrong with the last download using the link above.

Sorry, I cannot confirm this. I have tried two physical machines with Windows 10 and 11 and everything works as expected. ZIP archives don't support changing write permissions of the FS. I can only suppose that some anti-virus software prevent an execution of the software from unknown source in a such way and marks the directory as read-only. The #739 doesn't contain any FS interaction and cannot change the directory write permission

>  The latest download from the link below doesn't have this problem but unfortunately selection box and cursor positions are still wrong as shown below.

The latest continuous release doesn't contain #739 merged and the symbol editing is not fixed in it. 

>  As I go through old projects with subcircuits I'm finding more symbols that are "corrupted". 

This issue is the same issue with `QPainter`. Everything that is drawn after ID Text has a shift. This is fixed by #739. I don't observe the mentioned issues on my build anymore.

ra3xdh, 2024-06-09:
I am merging #739. Please try the next continuous release. It contains a fix for symbol editing.

closed 2024-06-09 by ra3xdh

tomhajjar, 2024-06-09:
I confirm the latest nightly release fixed the issue

commit bf6cfe42 (2024-06-09): Merge pull request #740 from wawuwo/739-fix-cursor-shift-in-symbol-mode

Fix #739: Restore QPainter state after paining the id_text element