ra3xdh/qucs_s #743: Windows Security Complaints

tomhajjar, 2024-06-09:
A long term issue has been Windows Security flagging Qucs-S as being from an "Unknown Publisher".

A newer issue is that when I copy the nightly release over existing files in the Qucs-S folder. It triggers multiple Security Scan warnings including trying to get me to do an "Online Scan".

[image]

ra3xdh, 2024-06-09:
> A long term issue has been Windows Security flagging the "first run" of Qucs-S as being from an "Unknown Publisher". 

The only way to fix this is to buy code signing certificate and sign the executable. I don't consider this option. 

> t triggers multiple Security Scan warnings including trying to get me to do an "Online Scan".

I have just tested a continuous release archive from the latest build and cannot confirm this. Everything works as expected except the "unknown publisher warning". I cannot say what triggers the security scan. Probably we have different UAC settings. I will also try to add stripping debug info step in the build script. Anti-virus software doesn't like debug binaries.

tomhajjar, 2024-06-09:
These issues might confuse new users. Maybe a warning label at the download location.

My UAC settings are always the lowest. The signing certificate has always been an issue so no big deal. The other warnings are fairly recent. Probably since the nightly releases. Windows Security steals all the resources and it can take a while before all the new files overwrite the existing files in the Qucs-S folder. What's stupid is Windows opens the Edge Browser, goes to an online-scan description page and doesn't seem to do anything. All the while "Antimalware" starts running in the background.

ra3xdh, 2024-06-23:
@tomhajjar Please check the archive at the following link: https://drive.google.com/file/d/1sUqvtKmHVVjeSqNP8h8GK-ijP8QvPQwX It is built manually and has executables with strupped debug info.

closed 2024-06-23 by ra3xdh

ra3xdh, 2024-06-23:
Thanks for the feedback. I will use the explicit debug info strip when preparing a release package. I have also added this step to the automated CI script for continuous release preparation. See #782. Please test if the security issues are gone in the latest continuous release.

tomhajjar, 2024-06-23:
MING??

[image]

dsm, 2024-06-23:
ucrt64 is fine for win10 and win11 and better than mingw64 and snapshot builds doesn't give warning for me on win11 and I didn't merge two different version. defender gives false positive and snapshot build less download count compare to release build, I think issue related to this and you merge qt5 and qt6 build into one.

tomhajjar, 2024-06-23:
I can overwrite files without issue.

Windows is still doing the first time opening warning for qucs-s.exe. Is this to be expected because of the lack of a security certificate?
[image]

ra3xdh, 2024-06-23:
Yes, the warning on the first start cannot be resolved without purchasing a security certificate. I don't consider this option.