ra3xdh/qucs_s #744: Schematic Editor Image Artefact

3813127458, 2024-06-09:
The Nutmeg script interferes with certain locations in the schematic:

https://github.com/ra3xdh/qucs_s/assets/75557552/04ff2f18-fc5f-445b-a76b-f5a5c962c12f

Schematic:
[BHBM_V4.zip](https://github.com/user-attachments/files/15754304/BHBM_V4.zip)

Qucs-S 24.2.0

Operating System: openSUSE Tumbleweed 20240607
KDE Plasma Version: 6.0.5
KDE Frameworks Version: 6.2.0
Qt Version: 6.7.1
Kernel Version: 6.9.3-1-default (64-bit)
Graphics Platform: Wayland

ra3xdh, 2024-06-10:
It is a known problem related to representation of a multiline text. I will try to provide a fix if possible. I have added a task list for this issue.

3813127458, 2024-06-10:
... to add, regarding Nutmeg: 

1. It should be possible to edit the Nutmeg script also directly in the schematic (without double clicking the Nutmeg script symbol), in order to be consistent with other blocks like .MODEL or .PARAM, where it is possible.
Both methods are useful.

2. The Nutmeg script parser should accept variables, eg.  {var1} , defined in a .PARAM block

Maybe you can put this on a feature list for future releases.

ra3xdh, 2024-06-10:
Multiline text bounding rect is already fixed by #723

ra3xdh, 2024-06-10:
> It should be possible to edit the Nutmeg script also directly in the schematic

This is not possible to implement in the near feature. The schematic editor uses `QLineEdit` control without border as the properties editor. This control doesn't support mulitline text. It's need to replace every occurrence of this editor by some other text editing control. I don't consider such a massive change because of one simulation.

> The Nutmeg script parser should accept variables, eg. {var1} , defined in a .PARAM block

It is not a straightforward task. The Ngspice postprocessor knows nothing about .PARAM and it's need to duplicate them in the `.control` section. The .PARAM may have incompatible syntax with Nutmeg script and it not possible to handle all cases automatically.

ra3xdh, 2024-06-11:
#747 implements blocking of the text editing of the Nutmeg script text.

closed 2024-06-11 by ra3xdh