ra3xdh/qucs_s #750: Issues Editing Sub-circuit Symbols

tomhajjar, 2024-06-11:
Still having issues editing and creating Sub-circuit Symbols. #739 didn't fix all the issues.

The small thin black "+" at 0,0 disappears during editing and gets replaced by a larger thick black "-". I then lose track of the "Centroid" for the symbol. I have to type Esc to get it to reappear but then I can't edit.

The Port labels ANODE/CATHODE are randomly disappearing, I can only get the the labels back if I go back to Schematic mode and then re-enter Symbol editing mode. I haven't figure out what sequence of editing steps causes this to happen.

[image]

[SOD323_pkg.zip](https://github.com/user-attachments/files/15793656/SOD323_pkg.zip)

wawuwo, 2024-06-12:
Hi! 

I've managed to reproduce two bugs and I'll try to provide fixes for them in the near future.

#### 1. A dash appears at 0,0 when anything on schematic is moved
  Steps to reproduce:
  1. Select any component from left panel and move cursor around schematic or using the mouse move any element already present on schematic

A small dash appears near the coordinates origin cross

#### 2. In symbol edit mode port labels disappear once you use the "Undo"
  Step to reproduce:
  1. Switch to symbol editing mode
  3. Ensure you have a port with label
  4. Do something (add a line for example)
  5. Hit Ctrl+Z
  
  Port labels disappear until switching to schematic mode and back to symbol mode.

<hr>

>if you select the entire diode symbol, move it and then use "CTRL R", the entire symbol disappears 

I couldn't reproduce this, sorry. Nothing disappears on my machine:

https://github.com/ra3xdh/qucs_s/assets/40355531/4392b12d-dd05-49a0-9fdc-5db111a874aa

>If I select an item and type "CTRL R" the item disappears. Doesn't matter if in schematic or symbol editing modes.

I cannot reproduce this one too. Both Mac and Linux machines don't have this problem:

https://github.com/ra3xdh/qucs_s/assets/40355531/516d2991-a692-42f4-a0b4-951abcf621e1

>If I type "CTRL R", Qucs-S is then "Stuck" in Rotate mode and I have to type "ESC".

I believe it has been always acting like this. I have the 24.1 version installed and hitting Ctrl+R in it "switches" you to "rotate mode", Ctrl+M to "mirror mode", etc.

tomhajjar, 2024-06-12:
I have to wait for the next nightly...

ra3xdh, 2024-06-13:
Fixed by #752 and #754.

tomhajjar, 2024-06-13:
Downloaded the latest nightly. 

If I zoom in on a sub-circuit symbol, select a group of items, move them and rotate them using CNTRL-R they disappear for good. Undo does nothing...

[image]

ra3xdh, 2024-06-13:
> If I zoom in on a sub-circuit symbol, select a group of items, move them and rotate them using CNTRL-R they disappear for good. Undo does nothing...

Sorry, I have checked both Linux and Windows versions and still cannot reproduce this. The rotation works as expected on my machine.

[Screencast_20240613_211848.webm](https://github.com/ra3xdh/qucs_s/assets/4920080/bc05ced4-b5f1-4f4f-ae0c-731cffa78837)

tomhajjar, 2024-06-13:
I can reproduce the same issue in Qucs 0.0.19 and 0.0.20. Not sure what to make of it.

wawuwo, 2024-06-13:
The only way to reproduce something similar I've found is to press Ctrl+R _while_ moving the elements.

The steps to reproduce are:
1. Select one or more elements on schematics or one or more paintings in symbol mode.
2. Begin moving them
3. Without releasing the left mouse button press Ctrl+R

The result:
1. "Rotation mode" turns on, cursor changes its appearance
1. Selected elements disappear completely and cannot be restored with the undo action

Is this the issue we're trying to find?

tomhajjar, 2024-06-13:
That's the issue I have been seeing. It seems like it has existed for over decade in Qucs as well. Seems odd no one saw it before now.

ra3xdh, 2024-06-14:
Yes, I understood how reproduce this now. All Qucs-S version are affected. The solution may be straightforward. It would be sufficient to disable rotate shortcut while moving symbol in symbol editing mode. I am reopening this.

ra3xdh, 2024-06-14:
I have added TODO list for this.

closed 2024-06-19 by ra3xdh