ra3xdh/qucs_s #759: New Varactor Library for ngspice

ra3xdh, 2024-06-16:
### Discussed in https://github.com/ra3xdh/qucs_s/discussions/758

<div type='discussions-op-text'>

<sup>Originally posted by **tomhajjar** June 16, 2024</sup>
Attached is a new library for ngspice called Varactor_RF.lib. Also attached is a project used to test the models and develop the library, Models include packaging parasitics. Should be used with the latest ngspice build that allows larger Vj and M.

[image]

[image]

[image]

[Varactor_ngspice_prj.zip](https://github.com/user-attachments/files/15855870/Varactor_ngspice_prj.zip)

</div>

tomhajjar, 2024-06-16:
The library doesn't include the BB814 dual varactor. I couldn't get a test fixture to work for the two diode in series. Single diode and two in parallel works but not in series. Capacitance should be 1/2 of one diode but it measured negative...

[image]

[image]

[image]

tomhajjar, 2024-06-17:
I got the BB814 dual varactor to simulate using S-parameters instead of the previous method. Unsure why the original method doesn't work for a dual varactor. 

Updated Varactor_RF.lib so please delete previous one in zip above.

[image]

[Varactor_ngspice_prj.zip](https://github.com/user-attachments/files/15860456/Varactor_ngspice_prj.zip)

ra3xdh, 2024-06-18:
Varactor_RF library has been uploaded by #767. I have slightly modified the library source, so it works now with both Qucsator and Ngspice.

tomhajjar, 2024-06-18:
I am almost done updating Varactor.lib. Attached sample is 95% done.

I added descriptions and changed Rs to reflect the real varactor. This library is OK as long as one recognizes it only works at maybe VHF frequencies and 95% of the diodes are obsolete. 

[Varactor.zip](https://github.com/user-attachments/files/15888807/Varactor.zip)

ra3xdh, 2024-06-18:
> I am almost done updating Varactor.lib. Attached sample is 95% done.

Thanks! Please open a new PR when you finish the library update. I will upload it in the repository.

tomhajjar, 2024-06-18:
Here is the finished library. 

[Varactor.zip](https://github.com/user-attachments/files/15889213/Varactor.zip)

ra3xdh, 2024-06-18:
> Here is the finished library.

I have added the updated `Varactor.lib` library to #767.

closed 2024-06-18 by ra3xdh