ra3xdh/qucs_s #765: Sub-circuit Text Displayed

tomhajjar, 2024-06-18:


ra3xdh, 2024-06-18:
To make the issue more clear, it is about graphics text object inside subcircuit drawing. The `ID_text` object is not affected. This may be another  consequence of #723

ra3xdh, 2024-06-18:
The library devices also seems to be affected. The issue was unnoticed, because the most of library devices don't use the rotated text inside symbol.

wawuwo, 2024-06-20:
Hi guys, sorry for the silence; it has been a bit stressful in recent days. I've prepared a fix for the problem. Please let me know if it breaks something else while fixing the problem :)

ra3xdh, 2024-06-21:
Fixed by #777

closed 2024-06-21 by ra3xdh

tomhajjar, 2024-06-21:
Font size is smaller when placed in the schematic than the original sub-circuit.

[image]
[image]

ra3xdh, 2024-06-21:
The font size requires an another fix. Please wait for the next patch. @wawuwo is aware about font size problems in some devices.

PR #777 (2024-06-21): Generalise Text::angle() logic
Hi!

I've rewritten `Text::angle()` to make it generic and able to handle any combination of `mCos` and `mSin` values. Before that `Text::angle()` could handle only three distinct cases and all other cases fell in "default", it was rather "stiff". I believe that is the reason why subcircuit symbol has text with wrong rotation angle.

It looks good now (at least on my machine):
[image]

May fix #765
commit c1702d3c (2024-06-21): Generalize Text::angle() logic

May fix #765