ra3xdh/qucs_s #769: Parameter Sweep of Varactor

tomhajjar, 2024-06-19:
I'm not having luck getting a varactor tuned BPF to work in ngspice. Works under Qucsator. I revisited the project in #768 looking for ideas but I couldn't get that project to work either.

I don't need to use a Nutmeg script to get this to work do I?

[image]
[image]

[VCO_prj.zip](https://github.com/user-attachments/files/15895275/VCO_prj.zip)

dwarning, 2024-06-19:
Something went wrong.
If I change in diagram properties v(vout) to ac.v(vout) the curves are as they should.

ra3xdh, 2024-06-19:
It is not a bug. You forgot to change the variables after changing the simulator. Probably you forgot to switch the simulator to Ngspice too. After switching the simulator to Ngspice and selection of the correct output variable name everything works as expected. Also keep in mind that Qucsator doesn't accept the source name as the sweep parameter name. The schematic requires the following modification to works with Ngspice/Qucsator seamless. 

[image]