ra3xdh/qucs_s #779: What is the rationale for the 0 dBm voltage value in power source?

ra3xdh, 2024-06-23:
### Discussed in https://github.com/ra3xdh/qucs_s/discussions/778

<div type='discussions-op-text'>

<sup>Originally posted by **csrabak** June 22, 2024</sup>
The magnitude of the voltage source for the Qucs-s power source, available in the components->sources pane puts 0.316228 V as magnitude for a 0 dBm power, 50 Ω (default impedance).

The line of the exported netlist is:

```
VP1 out 0 dc 0 ac 0.316228 SIN(0 0.316228 1MEG) portnum 1 z0 50
```

Since simulators use magnitude for this parameter, which is the _peak_ voltage of the source, this value would be OK for an open circuit, but due the internal impedance only one fourth of a milliwatt is available to the matched load.

Shouldn't the voltage be such that when a matched (i.e. 50 Ω) load is fed the power delivered to the load be 0 dBm (1 mW)?</div>

closed 2024-06-23 by ra3xdh

tomhajjar, 2024-06-23:
I forgot to show the data from ngspice was different than Qucsator...

[image]
[dBm_Voltage.zip](https://github.com/user-attachments/files/15944618/dBm_Voltage.zip)

tomhajjar, 2024-06-23:
Seems to be fixed in update I just got

[image]

ra3xdh, 2024-06-23:
Yes, the Ngspice/Qucsator data mismatch was fixed by #780. The latest continuous build has this fix.

PR #780 (2024-06-23): Fix Pac to match result from Ngspice and Qucsator
This PR fixes #779 and #778 I have corrected the equations to match result from Ngspice and Qucsator.