ra3xdh/qucs_s #784: Coupled transmission lines

DanielSank, 2024-06-24:
I am interested in circuits with coupled transmission lines. I find https://github.com/ra3xdh/qucsator_rf/blob/7494604ceab49ad18c5e25390078479add4594c1/src/components/ctline.cpp which has a model for coupled transmission lines. However, I do not see this model in the GUI when building a schematic.

How would I access this model in the schematic?

ra3xdh, 2024-06-24:
Everything is simple. You are simulating in Ngspice mode. The microstrip devices are not compatible with Ngspice and hidden to prevent its insertion in the schematic by mistake. You have to switch to Qucsator mode using the drop-down list in the toolbar. Make sure you are using the version >= 24.2.0 It is shipped with QucsatorRF by default. 

[image]

ra3xdh, 2024-06-24:
I have transferred this issue to discussions/Q&A https://github.com/ra3xdh/qucs_s/discussions You may ask the questions not related to bug reports here.