ra3xdh/qucs_s #786: Fill blue devices parameters from SPICE model

ra3xdh, 2024-06-24:
This is follow-up for #679. It would be good to add a possibility to convert the SPICE model to unified nonlinear device parameters from the GUI. I propose to add a special button like *Fill from SPICE model* in the component dialog. The user can press this button and insert the SPICE modelcard `.MODEL` code. The device properties will be filled automatically. This button should be available for the following devices:  Diode, BJT, MOSFET, JFET

tomhajjar, 2024-06-27:
How would you handle parameters not used by ngspice or extra info used by LTspice like "mfg=xxx". Would you just ignore these items?

Another issue is units used. Would you use 1e-9 or 1n as is? Leading 0? 0.01  versus .01

ra3xdh, 2024-06-28:
> How would you handle parameters not used by ngspice or extra info used by LTspice like "mfg=xxx"? 

The only option is to ignore such model parameters. Qucsconv also ignores them. 

> Another issue is units used. Would you use 1e-9 or 1n as is? Leading 0? 0.01 versus .01

It is possible to apply the units conversion. Qucs-S parser accepts both notation. It should be possible to implement a checkbox like *Convert number notaiton* in the model import dialog.

ra3xdh, 2024-07-03:
The proposed feature implemented by #795

ra3xdh, 2024-07-04:
Here is a new look of the component properties dialog and new SPICE model import window. 
[image]
[image]

ra3xdh, 2024-07-04:
The feature implemented by #795. Closing this.

closed 2024-07-04 by ra3xdh

tomhajjar, 2024-07-04:
If leading zero is missing does it add it?

ra3xdh, 2024-07-04:
No, it currently cannot handle missing leading zero. I am not sure if Ngspice allow such notation. I may add a patch implementing this conversion.

tomhajjar, 2024-07-04:
Another issue was some Spice models using the Greek µ instead of u. ngspice doesn't like µ

ra3xdh, 2024-07-04:
> Another issue was some Spice models using the Greek µ instead of u. ngspice doesn't like µ

I cannot do anything with this. Ngspice doesn't support Unicode.

tomhajjar, 2024-07-06:
I downloaded the latest and don't understand how to get this function to work. The Fill from SPICE Model is grey'd out. What is the procedure for loading the data or a file? 

Also shouldn't this be called Fill "Red" devices...

[image]
[image]

[2N5109.zip](https://github.com/user-attachments/files/16114419/2N5109.zip)