ra3xdh/qucs_s #789: Add OP variables dump feature

ra3xdh, 2024-06-26:
This feature was requested by @dwarning in discussion #761 It should be technically possible to dump the OP variables like @D1[cd] The following tasks should be done:

* The `@` symbol is reserved for `PlotVs` emulation in Ngspice mode. It's need to resolve this conflict.
* Add a special parameter for every simulation like `DumpOPVars=true/false`

viticleri, 2026-10-02:
Hi @ra3xdh,

I'm interested in working on this issue. Before starting the implementation,
I'd like to confirm two design points:

1. Option and scope. The `.DC` simulation already has a `saveOPs` property
   ("put operating points into dataset"), which is currently only passed to
   Qucsator. For Ngspice, would you prefer to reuse `saveOPs`, or introduce a
   separate option such as `DumpOPVars`?

   Also, would it be acceptable for the initial implementation to cover
   DC/OP, DC sweeps and parameter sweeps first, leaving AC/TRAN for a later
   extension?

2. Variable naming. Ngspice exposes these vectors using SPICE instance names;
   for example, a Qucs MOSFET `T1` is netlisted as `MT1`, so its
   transconductance would be referenced as `@mt1[gm]`.

   Would you prefer to:

   a) preserve the Ngspice name and adapt the PlotVs handling so that
      `@dev[param]` is not interpreted as PlotVs syntax, or

   b) rename these vectors while converting them to the Qucs dataset, for
      example `T1.gm`, which matches how Qucsator names operating points when
      `saveOPs=yes`?

In either case, I would keep the feature opt-in and disabled by default, as
discussed in #761.

Thanks!

ra3xdh, 2026-10-16:
> For Ngspice, would you prefer to reuse saveOPs, or introduce a separate option such as DumpOPVars?

Yes, an explicit option would be preferable. 

> Also, would it be acceptable for the initial implementation to cover DC/OP, DC sweeps and parameter sweeps first

Yes, this would be acceptable

> Variable naming. 

A simple solution may be to drop the leading `@` symbol when such variable is found in the dataset.