ra3xdh/qucs_s #790: Bug: DC simulation does not work

3813127458, 2024-06-29:
DC simulation does not work, because DC simulation block is not included in the netlist.

[image]

[Interlock_prj.zip](https://github.com/user-attachments/files/16039750/Interlock_prj.zip)

* Qucs 24.2.1  /QUCS-S/Interlock_prj/ilk1.sch
.INCLUDE "C:/Program Files (x86)/Qucs-S/share/qucs-s/xspice_cmlib/include/ngspice_mathfunc.inc"
R3 0 _net0  10K tc1=0.0 tc2=0.0 
R2 _net0 _net1  47K tc1=0.0 tc2=0.0 
R1 _net1 _net2  1M tc1=0.0 tc2=0.0 
R4 _net3 _net4  470 tc1=0.0 tc2=0.0 
R5 _net4 _net2  1M tc1=0.0 tc2=0.0 
V1 _net3 0 DC 12

.control

op
print   > spice4qucs.dc1.ngspice.dc.print
destroy all
reset

exit
.endc
.END

ra3xdh, 2024-06-29:
It is not a bug. The DC black function has changed since v24.1.0 See the #463 and #124 for detailed explanation. Not all new features are documented. The single DC block is mapped to OP now. The DC bias mode (F8) works as expected. The DC simulation is mapped to OP and exports volatages to the datset only for labeled node. Add at least one node label to see the data in dataset. Add the Parameter Sweep block to make DC sweep from DC block. 

[image]

closed 2024-06-29 by ra3xdh

3813127458, 2024-06-29:
right, ... I did not draw a node label. This was the reason why Ngspice crashes. 
With at least one node label everything works fine.

ra3xdh, 2024-06-29:
> This was the reason why Ngspice crashes.

Ngspice didn't crash on my machine on this schemaric. It only makes an empty dataset.

3813127458, 2024-06-30:
Probably it is a special case:
on Windows and by using an external Ngspice kernel:

[image]

the simulator never finish the DC simulation in my case:

[image]

and after stop manually the never ending simulation there is a Ngspice error note:
[image]

Overall it is just a cosmetic issue but probably the never ending simulation could be handled somehow.