ra3xdh/qucs_s #794: MacOS installed qucs-s not working

bnilsson11, 2024-07-02:
I'm trying to build and install qucs-s 24.2.1 from source.
The installed version (/usr/local/bin/qucs-s.app) is not working under MacOS Version 14.5 (23F79).

process:

git clone https://github.com/ra3xdh/qucs_s

Building the source using

cd qucs-s
mkdir builddir
cd builddir
cmake ..
make
make install

runs without errors.
I am using Xcode and MacPorts for the "infrastructure" including cmake and Qt5.

However, the installed /usr/local/bin/qucs.s.app exits immediately.
Starting from terminal in qucs-s.app/Contents/MacOS/ gives:
@MacStudio MacOS % ./qucs-s
Warning: Could not find the Qt platform plugin "cocoa" in "" (:0, )
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
 (:0, )
zsh: abort      ./qucs-s

The installed qucs-s.app contains a Frameworks folder with the necessary libraries for standalone operation including the Qt libs. The built/not_yet_installed qucs-s.app in qucs-s/builddir/qucs does not have this Frameworks folder and runs without problems.
I assume the deployment process fails in some way.

I have applied a workaround by copying the built app without Frameworks to /usr/local/bin and use this instead.
So far it seems to work, but there may be some things not working that I have not discovered yet.

ra3xdh, 2024-07-02:
Unfortunately I cannot suggest anything useful for Mac, because I have no access to Mac hardware and not using this platform. @dsm is using Mac as the development platform and may suggest how to build and run application on Mac. The answer may take a while. Please wait. 

@dsm, Could you suggest something? Maybe a some obvious step is missing.

dsm, 2024-07-02:
I'm able to building and running latest version of qucs_s on macos 14 arm64 version.

I will sent .app file to as soon as possible to accesssing my macbook and I will improve macos ci to release mac version.

I have one issue all qucs tools install to Application folder but not able to run inside qucs_s ex: qucs activefilter issue related to calling path wrong changing path solve this issue or combine all exe to one .app file how should we solve I am not sure.

dsm, 2024-07-02:
I'm not using make install I just copy the .app to Application folder and I am building with homebrew-qt6

dsm, 2024-07-02:
can you try this it includes all tools and built with qt 6.7.0, I use `make install DESTDIR=./deploy` and it stop with error and .app file don't run and I run `/opt/homebrew/Cellar/qt/6.7.0_1/bin/macdeployqt qucs-s.app -always-overwrite` and everything works fine and it doesn't find examples folder I solve this later.  

[qucs_latest.zip](https://drive.google.com/file/d/1ZOBiqo_1fIdFEO4iaYZi-U6Td8MqIaif/view?usp=sharing)

dsm, 2024-07-02:
after installing qucs-s, please run this command: `xattr -c /Applications/qucs-s.app`

bnilsson11, 2024-07-02:
I built the latest qucs-s earlier today.
I followed your advice but for my MacPorts environment:

cd qucs-s/builddir/qucs
/opt/local/libexec/qt5/bin/macdeployqt qucs-s.app

and now all is working.
But as you comment, there are no examples.
So, this issue is partially solved.

dsm, 2024-07-02:
I didn't try macports, our CI use homebrew. Can you try provided dmg file to install qucs_s. 

Refactoring needed some section of the cmake file for deploying mac app without an issue. 

Maybe qucs_s tools install seperatly in a folder inside /Applications folder, we should found a optimal way to this. 

App don't run my machine to download provided dmg file without xattr, I'll solve this and update mac os CI to finally we have macos release every commit.

Issue should still remain open. I'll try to solve all.

bnilsson11, 2024-07-02:
I tried to run the app it as-is, and my system reported that it is damaged and should be put in the bin.
Then I did "xattr -c qucs-s.app".
I tried to run it, but this time it exited unexpectedly.
Crash report says:
Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x000000010186c000
Exception Codes:       0x0000000000000032, 0x000000010186c000

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page

dsm, 2024-07-02:
can you try this `codesign --force --deep --sign - /Applications/qucs-s.app` and this `xattr -dr com.apple.quarantine /Applications/qucs-s.app`

dsm, 2024-07-02:
I am installing a virtualmachine to test issues and I will provide a solution.

dsm, 2024-07-02:
I tried this two in a virtual machine and all works, can you try this `codesign --force --deep --sign - /Applications/qucs-s.app` and this `xattr -cr /Applications/qucs-s.app`

dsm, 2024-07-02:
the toolbar may be invisible on first startup, it is fixed when you close and open it from the view menu.

bnilsson11, 2024-07-03:
Now it works, thank you very much.

> 
> 
> I tried this two in a virtual machine and all works, can you try this codesign --force --deep --sign - /Applications/qucs-s.app and this xattr -cr /Applications/qucs-s.app

Bengt Nilsson
Ansåsvägen 8
43853 Hindås
Sweden
***@***.***

ra3xdh, 2024-07-03:
Closing as completed. Thanks @dsm !

closed 2024-07-03 by ra3xdh

bnilsson11, 2024-07-03:
Did you try “make install” successfully?MvhBengt NilssonSkickat från min iPhone3 juli

dsm, 2024-07-03:
make install not working properly, some cmake files not prepare good, I think make install should be use deploy purpose. now I am preparing Macos Release via github actions and I manually packed app bundle, I have to solve some minor issue and maybe make install removed for macos.

bnilsson11, 2024-07-03:
If you make a macos release bundle, please take a look at “About qucs-s”.Now it only shows the mandatory Qt info. There is nothing about which qucs-s version is running.The standard way would be one “about Qt” and one “about qucs-s”.As far as I can tell, there is no way now to know which version you have.MvhBengt NilssonSkickat från min iPhone

dsm, 2024-07-03:
thank you I look into this. can you test this ci build. [qucs_s_macos_arm64](https://github.com/dsm/qucs_s/actions/runs/9776381750/artifacts/1663466625)

bnilsson11, 2024-07-03:
The app opens ok, but the system device libraries are missing if the app is placed in /Applications or anywhere else.
The system libraries become visible again if the app is placed in /usr/local/bin.

The libs are placed in /usr/local/share/qucs-s/library on my system.
If this is the correct location I cannot see how ”make install” can be omitted.

BTW, app ”Overview” shows the version info.
I discovered this by chance, not the expected method to get the usual app info.

> 
> 
> thank you I look into this. can you test this ci build. qucs_s_macos_arm64 <https://github.com/dsm/qucs_s/actions/runs/9776381750/artifacts/1663466625>

Bengt Nilsson
Ansåsvägen 8
43853 Hindås
Sweden
***@***.***

bnilsson11, 2024-07-03:
Compare the old qucs, where About.. is designed in the standard way.

￼
￼

> 
> 
> thank you I look into this. can you test this ci build. qucs_s_macos_arm64 <https://github.com/dsm/qucs_s/actions/runs/9776381750/artifacts/1663466625>

Bengt Nilsson
Ansåsvägen 8
43853 Hindås
Sweden
***@***.***

dsm, 2024-07-03:
#797 is a dedicated issue for macos improvements we can continue there.