ra3xdh/qucs_s #796: Where cmake is looking for functions?

dwarning, 2024-07-03:
Building since years successful numerical applications on my laptop I am surprised that my own build for qucs-s is working very well if I see the excerpt output of the cmake command to stdout:

-- Looking for floor
-- Looking for floor - not found
-- Looking for pow
-- Looking for pow - not found
-- Looking for exp
-- Looking for exp - not found
-- Looking for sqrt
-- Looking for sqrt - not found
-- Looking for log10
-- Looking for log10 - not found
-- Looking for log
-- Looking for log - not found
-- Looking for cos
-- Looking for cos - not found
-- Looking for sin
-- Looking for sin - not found
-- Looking for acos
-- Looking for acos - not found
 and much more.
This is under ubuntu 22.04.

ra3xdh, 2024-07-03:
I am getting the same warning on all platforms, but everything could be built without problems. Probably the Cmake checks for some specific function versions. I think these warning may be ignored.

dsm, 2024-07-03:
this checks related to qucsator_rf

ra3xdh, 2024-07-04:
@dwarning , See the https://github.com/ra3xdh/qucsator_rf/issues/10 for the follow-up for this.