ra3xdh/qucs_s #797: Improve MacOS support.

dsm, 2024-07-03:
Hi, I am preparing MacOS CI for snapshot release. 

this issue was created to keep track of improvements and issues in one place. I will add items later.

dsm, 2024-07-03:
@ra3xdh  In qucs-init.cpp start line after 666 disabled for macos and About qucs-s run as expected but why I didn't understand. 

```
#ifdef __APPLE__
  helpAboutQt = new QAction(tr("About Qt..."), this);
  helpAboutQt->setStatusTip(tr("About Qt"));
  helpAboutQt->setWhatsThis(tr("About Qt\n\nAbout Qt by Trolltech"));
  connect(helpAboutQt, SIGNAL(triggered(bool)), qApp, SLOT(aboutQt()));
#endif
```
I also disabled this in the same file.
```
#ifndef __APPLE__ 
  helpMenu->addAction(helpAboutQt);
#endif
```

dsm, 2024-07-03:
I almost finished new macos ci, About qucs-s dialog fixed. and examples, library and symbols folder added. Please test this release and give feedback, thanks. 

[qucs_s_24.2.1_arm64_.dmg](https://github.com/dsm/qucs_s/releases/download/continuous_build/qucs_s_24.2.1_arm64_.dmg)

ra3xdh, 2024-07-04:
Thanks! Let's wait for the feedback, then please prepare a patch.

dsm, 2024-07-04:
patch is on the way, can you update qucsator_rf submodule.

ra3xdh, 2024-07-04:
I have just updated qucsator_rf by this commit https://github.com/ra3xdh/qucs_s/commit/6e6fb4fb70fd90d3ed8e9b171fb6673f95ceb222

dsm, 2024-07-04:
#799 fixes prepared.

closed 2024-07-04 by ra3xdh

bnilsson11, 2024-07-04:
Sorry, I cannot find [qucs_s_24.2.1_arm64_.dmg](https://github.com/dsm/qucs_s/releases/download/continuous_build/qucs_s_24.2.1_arm64_.dmg)
I get a 404 when clicking on it.
Where should I look?

dsm, 2024-07-04:
macos release patch is merge to main repo you can look to release section.

bnilsson11, 2024-07-04:
Ok, got it.
About.. is fine.
No System Libraries.

dsm, 2024-07-04:
I don't know, when I tried yesterday there is no problem with system libraries

dsm, 2024-07-04:
sorry in ci there should be mistakenly change folder name 'share' to 're' and didn't work

bnilsson11, 2024-07-04:
True. I changed `re` to `share` and now it is working.

dsm, 2024-07-04:
latest commit I changed mistakenly :) this issue fixed in #800.