ra3xdh/qucs_s #798: File based s-parameter simulation with Xyce

dwarning, 2024-07-03:
I tried following setup:
[image]
Simulation runs without complain - but results show nothing.
Files have this format (which are working with ngspice):

`!NPN
!s-parameters at Vbe=0.78V Last updated 17/05/01

# HZ S RI R 50
! Freq         S11           S21            S12            S22
  1e+008          0.996556        -0.00612777     -2.20782        -0.00403262     1.70798e-005    0.00126266      0.998838        -0.00452896
  2e+008          0.997525        -0.01188        -2.16387        0.00692958      -4.00192e-007   0.00235935      0.999477        -0.00896073
`

andresmmera, 2024-07-03:
Hi,

Does it work when you simulate with qucsator?
Does your s2p file cover the whole frequency sweep?

ra3xdh, 2024-07-03:
It is not a bug. The touchstone file is mapped to `YLIN` device. This device works only with HB simulation. See the section 2.3.24 of the Xyce Reference guide and #452

closed 2024-07-03 by ra3xdh

dwarning, 2024-07-03:
@andresmmera : The s2p file covers the entire frequency range - I show only an excerpt.
@ra3xdh : Do you know an alternative for ac simulations?

ra3xdh, 2024-07-03:
It seems Xyce doesn't support the direct simulation of the touchstone file for SP and AC. I didn't find anything in the documentation. And there is no Xyce alternative for `s2spice` utility.