ra3xdh/qucs_s #804: [MacOS] add Qucs-S to homebrew package manager

dsm, 2024-07-05:
Hi, 

It would be good to add qucs-s to homebrew.

https://docs.brew.sh/Adding-Software-to-Homebrew#casks

dsm, 2024-07-08:
@ra3xdh  can you look at the ngspice homebrew build for macos configuration is OK for qucs-s.

[ngspice.rb](https://github.com/Homebrew/homebrew-core/blob/8c5fd91c3ad47aa2c7e83e24e875602c08721625/Formula/n/ngspice.rb)

ra3xdh, 2024-07-08:
Yes, the ngspice build configuration contains all necessary options set.

dsm, 2024-07-08:
I'll add qucs-s continuous to homebrew below is the draft.

```ruby
cask "qucs-s@nightly" do
  version "24.2.1"
  arch arm: "arm64", intel: "x86_64"

  url "https://github.com/ra3xdh/qucs_s/releases/download/continuous_build/Qucs-S-#{version}-macOS-#{arch}.dmg",
      verified: "github.com/ra3xdh/qucs_s/"

  name "Qucs-S"
  desc "Quite Universal Circuit Simulator with SPICE"
  homepage "https://ra3xdh.github.io/"

  depends_on formula: "ngspice"

  livecheck do
    url :url
    regex(%r{href=.*?/Qucs-S[._-]v?(\d+(?:\.\d+)+)-macOS-(arm64|x86_64)\.dmg}i)
  end

  app "Qucs-S.app"

  zap trash: [
    "~/Library/Application Support/qucs-s",
    "~/Library/Preferences/com.example.qucs-s.plist",
    "~/Library/Saved Application State/com.example.qucs-s.savedState",
  ]
end

```

dsm, 2024-07-17:
I created this PR to solve this issue, https://github.com/Homebrew/homebrew-cask/pull/179918

dsm, 2024-07-17:
this is the final code for to add cask to brew but macos on arm binaries must be sign and notarize so we need apple developer account but no one can afford this $99 fee so issue should be close.

```ruby
cask "qucs-s@nightly" do
  arch arm: "arm64", intel: "x86_64"

  version "24.2.1"
  sha256 :no_check

  url "https://github.com/ra3xdh/qucs_s/releases/download/continuous_build/Qucs-S-#{version}-macOS-#{arch}.dmg",
      verified: "github.com/ra3xdh/qucs_s/"
  name "Qucs-S"
  desc "Quite Universal Circuit Simulator with SPICE"
  homepage "https://ra3xdh.github.io/"

  livecheck do
    url :url
    strategy :github_latest
  end

  depends_on formula: "ngspice"

  app "qucs-s.app", target: "Qucs-S.app"

  zap trash: [
    "$HOME/Library/Application Support/qucs-s",
    "$HOME/Library/Preferences/com.example.qucs-s.plist",
    "$HOME/Library/Saved Application State/com.example.qucs-s.savedState",
  ]

```

ra3xdh, 2024-07-18:
@dsm  Thanks for effort. I don't consider buying Apple developer certificate to sign the binaries. We can try the approach using own tap. I can create an empty repository for example `qucs_s_tap` at my `ra3xdh` account. Then you will create a PR to upload a necessary files. After the tap setup has finished I will update the installation instructions at the Qucs-S website.

ra3xdh, 2024-07-18:
@dsm I have just created https://github.com/ra3xdh/homebrew-qucs-s and invited you as collaborator. You may send PR or use direct push to populate this repository. Then please send me an instruction about MacOS installation for placing on Qucs-S website. You may also make a PR to https://github.com/ra3xdh/ra3xdh.github.io/ with MacOS install instruction.

dsm, 2024-07-18:
I accepted your invitation thanks,  I'll test and push the necessary things.

dsm, 2024-07-18:
I updated homebrew-qucs-s repo so macos brew installation is succesfull

ra3xdh, 2024-07-19:
I have updated the installation instructions on the website. Closing this as completed.

closed 2024-07-19 by ra3xdh