ra3xdh/qucs_s #808: Not possible to clear pin assignment for SpiceLibComp on MacOS

ra3xdh, 2024-07-05:
This issue is related to MacOS only. Linux and Windows are not affected. It is not possible to clear pin assignment for new SpiceLibComp *Spice library device* after redesign #679. See #792 for context. Here are two screencasts illustrating the table behavior on MacOS and Linux. It seems that `QTableWidget::setCellWidget` works differently on different platforms. I am unable to fix this issue myself, need to wait for someone using Mac. 

*Linux (correct)

[Screencast_20240704_200738.webm](https://github.com/ra3xdh/qucs_s/assets/4920080/dcbc3eb3-c4d1-4075-a56f-bc88b806ee7d)

* MacOS wrong

https://github.com/ra3xdh/qucs_s/assets/4920080/355053d2-9549-4f44-8e20-8a523ff3d80f

dsm, 2024-07-05:
I will tried to fixed this issue. I download the same schematic lm335 symbol pin selection is behaving like a text box. selecting all NC give 4 pin but sometimes I double clicked text become 1NC and 1 is became invisible for the other selectin box. I will upload a video later.

dsm, 2024-07-07:
windows build also affacted.

[Windows_test.webm](https://github.com/ra3xdh/qucs_s/assets/41967334/fe18a1ba-bef1-404d-bba8-b07bfb88c96f)

dsm, 2024-07-07:
I tested on Ubuntu 24.04 wsl2 also same issue. 

[linux_wsl2.webm](https://github.com/ra3xdh/qucs_s/assets/41967334/172c4f64-4d15-4928-a933-108310d4d670)

ra3xdh, 2024-07-07:
Yes, I understood how to reproduce this. 

* Click on cell, drop-down list appears. 
* Don't select anything and double click on the next cell
* The drop down list remains to hang in the previous cell

If you select NC from the drop-down list, it disappears and releases pin.

ra3xdh, 2024-07-08:
Please cjeck #824 for the fix

dsm, 2024-07-08:
I tested windows and works as expected. I'll test macOS as soon as possible.

dsm, 2024-07-08:
this is macos test pin selection OK but some renderin artifact ex 1NC or N(empty_symbol)C

https://github.com/ra3xdh/qucs_s/assets/41967334/e7e68f24-b39e-4303-9b70-f8ad522be5d4

dsm, 2024-07-08:
[image]

ra3xdh, 2024-07-08:
I don't know why MacOS inserts random unicode symbols after "N". I didn't add anything in the item string. Maybe Qt has registered some auto-conversion for "NC" string. What is your locale setting?

dsm, 2024-07-08:
English(US) I think it writes NC twice but starting on C N(CN)C

ra3xdh, 2024-07-08:
Please try the following patch. It clears item text before `setCellWidget` call.
~~~
diff --git a/qucs/extsimkernels/spicelibcompdialog.cpp b/qucs/extsimkernels/spicelibcompdialog.cpp
index d028918f..cc1dd96a 100644
--- a/qucs/extsimkernels/spicelibcompdialog.cpp
+++ b/qucs/extsimkernels/spicelibcompdialog.cpp
@@ -388,6 +388,7 @@ void SpiceLibCompDialog::slotTableCellDoubleClick()
       cbxSelectPin->addItem(QString::number(i));
     }
   }
+  tbwPinsTable->item(r,c)->setText("");
   tbwPinsTable->setCellWidget(r,c,cbxSelectPin);
   connect(cbxSelectPin,SIGNAL(activated(int)),this,SLOT(slotSelectPin()));
~~~

dsm, 2024-07-08:
I am not accessing my mac right now but I'll try. (about 5 hours)

dsm, 2024-07-08:
@ra3xdh Hi, I tested above patch and it's working as expected.

ra3xdh, 2024-07-08:
Fixed by #824. Closing as completed.

closed 2024-07-08 by ra3xdh