ra3xdh/qucs_s #811: Deprecate *.qucs packages

ra3xdh, 2024-07-06:
The main application is able to pack the project contents to an archive having `*.qucs` extension. In fact the `*.qucs` files have self-designed archive format without compression. I don`t see any reasons to keep this homemade archiver in the future versions. Such approach may have many flaws, and nonstandard binary archives may be blocked by mail systems. This makes difficult to share such packages. So, I would remove this feature in the next release. Is anyone still use `*.qucs` files?

ra3xdh, 2024-07-06:
I propose something like right click on project in the left panel and then `Create ZIP` using system-wide ZIP utlity. KiCAD has similar feature.

tomhajjar, 2024-07-07:
Are you going to keep the function for legacy archives? The only ones I know of are the very old Qucs project examples. Some of these examples don't work for Qucs-S since they are Verilog.

ra3xdh, 2024-07-07:
I am planning totally remove the code related to this self-made archiver and provide a small external utility if someone will need to unpack old `*.qucs` archive.

ra3xdh, 2024-07-09:
The creating/extracting of the `*.qucs` packages deprecated by #831. The related archiver code has been removed from source tree. I will provide an utility to unpack `*.qucs` archives later.

closed 2024-07-09 by ra3xdh