ra3xdh/qucs_s #812: schematic error causes component line format error

bobu01, 2024-07-06:
qucs-s can write a schematic file that can't be opened later.  The example schematic has a spice .PARAM block. The block has a name string that contains unwanted space characters. The simulation runs fine and the schematic is saved without complaint. Later, the schematic fails to open due to the space characters in the .PARAM line. 

For example, this line is no good:
`  <SpicePar VCC and VREF 1 820 360 -27 18 0 0 "VCC=5" 1 "THH=VCC*2/3" 1 "THL=VCC/3" 1>`

[image]

This issue can be fixed by opening the .sch file in a text editor and removing the unwanted spaces from the name. (If you can find them.)

The system is an up to date Arch Linux distro. qucs-s version is 24-2-1, installed from Arch AUR.

[trig_load_sch.zip](https://github.com/user-attachments/files/16117060/trig_load_sch.zip)

tomhajjar, 2024-07-07:
So the .PARAM name was invalid. It was "VCC and VREF" versus "VCC_and_VREF".

What other characters beside a "space" can't be used?

ngspice doesn't like netlist filenames with "space" either.

ra3xdh, 2024-07-07:
The spaces in device name should not be allowed. I prepared a fix for this. See #814.

PR #814 (2024-07-07): Set validator for component name
This PR fixes #812. The sapces in component name are not allowed anymore.
closed 2024-07-07 by ra3xdh