ra3xdh/qucs_s #818: Dark theme issue on win11

dsm, 2024-07-07:
Hi, Qt added new theming for win11 named Windows11 style beginning Qt 6.7. This theme can automatically switch dark and light mode. In dark mode editing component color of the value is white. And some rendering artifact. I add some pictures.

Problem 1:

[image]

Problem 2:

[image]

Problem 3:

[image]

ra3xdh, 2024-07-07:
I have also recently noticed that the dots on device pins are disappeared after schematic rendering refactoring. 
* Old 
[image]
* New -- no dots
[image]

wawuwo, 2024-07-07:
Hi guys!

Here is a checklist of problems to check that I've understood them well and to control the progress. Please correct me if I'm wrong:

- [ ] When property value is being edited, value text is misplaced
- [ ] When property value is being edited and QT theme is switched to dark mode, value text has white color and is barely visible
- [ ] When property value is being edited, value text may is visible only partially
- [ ] Connections to component ports are rendered incorrectly; if port has only one connection is has to be rendered as a little filled square

dsm, 2024-07-07:
The application is already in dark theme, but when the value is changed, there is a white background and white text in windows11, but in macOS, this does not happen in the dark theme, it is white text on a black background. In problem 2, there is also a residue of the value that normally appears on the white background, which should not be there.

ra3xdh, 2024-07-08:
The #820 resolves the problem with nodes rendering. I have added TODO list for this issue.

wawuwo, 2024-07-08:
@dsm I made a patch which uses different approach to control colours when property values is being edited, but unfortunately I don't have a windows-based machine to try it out. Could you please try the patch to check if it fixes the problem №2?

dsm, 2024-07-08:
Yes this patch is fixed Problem 2.

ra3xdh, 2024-07-14:
The full fix provided by #842 Closing as completed.

closed 2024-07-14 by ra3xdh