ra3xdh/qucs_s #819: parameters not working in rectangular voltage source

bobu01, 2024-07-07:
In the rectangular voltage source component, some values don't work when using a parameter value.  U, U0, Tr, Tf, Td do work.  TH and TL do not work.  No error message is generated. A parameter like {value} seems to cause a zero pulse width in the .cir netlist. There are similar issues with T1 and T2 in the voltage pulse component. 

Please see the zip file for examples.  [rect_src.zip](https://github.com/user-attachments/files/16120139/rect_src.zip)

System is an up to date Arch Linux distro. qucs-s version is 24.2.1

ra3xdh, 2024-07-07:
It is not a bug. The rectangular voltage source has different definition of parameters in different simulator. The SPICE requires PW=TH+TL, but Qucsator requires TH and TL separately. Qucs-S GUI have to calculate the PW for SPICE from TH and TL. This cannot be done without conversion of the parameters in numeric form. So the variables in TH and TL are not allowed for SPICE mode.  Use the red V source and define the full SPICE string if you need variables.

ra3xdh, 2024-07-07:
I have made some tests and found that Ngspice accepts expressions inside the `PULSE` source definition. So, the variables for TH and TL parameters could be allowed. I will a fix for this soon.

ra3xdh, 2024-07-08:
Fixed by #825 
[image]

PR #825 (2024-07-08): Allow variables in Vrect parameters
This PR fixes #819 The variables for VRect source are allowed now.
closed 2024-07-08 by ra3xdh