ra3xdh/qucs_s #821: Rotate shortcut not working correctly.

dsm, 2024-07-07:
Rotate shortcut ctrl+r or cmd+r doesn't work correctly. Clicking a component ex. Ground symbol and pressing ctrl+r multiple times, component moves around on schematics. And also simulation box not rotating but moving around also. Also rotating component with shortcut makes component off the grid. Some components ex. Resistor it doesn't happen.

ra3xdh, 2024-07-08:
> Clicking a component ex. Ground symbol and pressing ctrl+r multiple times, component moves around on schematics.

I believe the problem with rotating of the 1-port devices was always present in Qucs-S, but remain undiscovered. It is surely not introduced by recent refactoring. I propose to keep the things as is. 

> And also simulation box not rotating but moving around also. 

The rotation of the simulation boxes and equations should be forbidden. It's need to add a check in rotate handler. 

> Also rotating component with shortcut makes component off the grid. 

Also there is an old problem. Should be fixed in v2.1.0

ra3xdh, 2024-07-08:
Device falling out of out grid after rotation should be fixed by #399

dsm, 2024-07-08:
there is a difference with select and press ctrl+r and press ctrl+r and press component.

[rotation.webm](https://github.com/ra3xdh/qucs_s/assets/41967334/ffbf444c-b4c4-45bf-b71b-7ce3157c737d)

gslg-1, 2024-07-13:
I have code that could solve the issue. But since I am very new to this project, please check carefully. 
[Screencast from 13.07.2024 13:34:04.webm](https://github.com/user-attachments/assets/61a40d1b-4db1-4921-b5a1-6d92ce31892b)
Since I am not able to push (I guess I need to prove myself ) I will just add the code snippet here.

**Update Schematic.ccp:**
void Schematic::relativeRotation(int &newX, int &newY, int comX, int comY, int oldX, int oldY)
{
    // Shift coordinate system to center of mass
    // Rotate
    // Shift coordinate system back to origin
    newX = (oldY-comY)+comX;
    newY = -(oldX-comX)+comY;
}

bool Schematic::rotateElements()
{
    Wires->setAutoDelete(false);
    Components->setAutoDelete(false);

    // To rotate a selected area its necessary to work with half steps
    int x1 = INT_MAX, y1 = INT_MAX;
    int x2 = INT_MIN, y2 = INT_MIN;
    QList<Element *> ElementCache;
    copyLabels(x1, y1, x2, y2, &ElementCache); // must be first of all !
    copyComponents(x1, y1, x2, y2, &ElementCache);
    copyWires(x1, y1, x2, y2, &ElementCache);
    copyPaintings(x1, y1, x2, y2, &ElementCache);
    if (y1 == INT_MAX)
    {
        return false; // no element selected
    }
    int comX = (x1 + ((x2-x1) / 2)); // center of mass
    int comY = (y1 + ((y2-y1) / 2)); 
    int newPosX = INT_MIN; 
    int newPosY = INT_MIN;

    Wires->setAutoDelete(true);
    Components->setAutoDelete(true);
    setOnGrid(comX, comY);

    Wire *pw;
    Painting *pp;
    Component *pc;
    WireLabel *pl;
    // re-insert elements
    for (Element *pe : ElementCache)
        switch (pe->Type) {
        case isComponent:
        case isAnalogComponent:
        case isDigitalComponent:
            pc = (Component *) pe;
            pc->rotate(); //rotate component !before! rotating its center
            relativeRotation(newPosX, newPosY, comX, comY, pc->cx, pc->cy);
            pc->setCenter(newPosX, newPosY);
            insertRawComponent(pc);
            break;

        case isWire:
            pw = (Wire *) pe;
            relativeRotation(newPosX, newPosY, comX, comY, pw->x1, pw->y1);
            pw->x1 = newPosX;
            pw->y1 = newPosY;
            relativeRotation(newPosX, newPosY, comX, comY, pw->x2, pw->y2);
            pw->x2 = newPosX;
            pw->y2 = newPosY;

            pl = pw->Label;
            if (pl) {
                x2 = pl->cx;
                relativeRotation(newPosX, newPosY, comX, comY, pl->cx, pl->cy);
[... 149 lines cut ...]
+                pl->x1 = pl->y1 - y1 + x1;
+                pl->y1 = x1 - x2 + y1;
             }
-            relativeRotation(newPosX, newPosY, comX, comY, pl->cx, pl->cy);
-            pl->cx = newPosX;
-            pl->cy = newPosY;
+            x2 = pl->cx;
+            pl->cx = pl->cy - y1 + x1;
+            pl->cy = x1 - x2 + y1;
             insertNodeLabel(pl);
             break;
 
diff --git a/qucs/schematic.h b/qucs/schematic.h
index 491deec4..5f0e75f0 100644
--- a/qucs/schematic.h
+++ b/qucs/schematic.h
@@ -370,7 +370,6 @@ private:
   void drawPostPaintEvents(QPainter* painter);
   void paintFrame(QPainter* painter);
   void drawGrid(QPainter* painter);
-  void relativeRotation(int &x, int &y, int comX, int comY, int posX, int posY);
 
 /* ********************************************************************
    *****  The following methods are in the file                   *****
(END)

ra3xdh, 2024-07-13:
@gslg-1 Thanks for providing a fix! I will test this in the next few days.

> Since I am not able to push (I guess I need to prove myself ) 

The best way for contribution is to make a pool request (PR). You don't need to be repository member to make pull request. This feature is available for everyone having Github account. You may look at this guide how to make a pull request https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

gslg-1, 2024-07-13:
I've provided a fork, with the changes and will create a pull request

ra3xdh, 2024-07-13:
Fixed by #840 Closing as completed.

closed 2024-07-13 by ra3xdh